microsoft/maker.js #57: Group or Layering Capability

zrichmond, 2015-09-05:
I have only begun to work with maker.js, which I have to say really fills a niche that has been missing from the javascript toolset for some time.  So many thanks for publishing such a great toolset.  However, one feature it seems to be missing is the concept of groups (svg) or layers (dxf).  The ability to set these, is important in CNC machining in order to  map tool paths to specific tools and other cut and tool specifics.  Although I see a few places to start, being so new to the codebase, I am not sure what the best approach would be.  I'd be inclined toward adding a group property to the path and model objects. Such that went the model group is set, contained paths would also be set. Then depending on the filetype being exported either group or layer would be named appropriately.  

If I could either be pointed in the right direction, or maybe this is a planned feature, which would be great, let me know.  Either way, even though I'm still pretty green with javascript, I've submitted a CLA and forked just incase we get to the point of a pull request.

zrichmond, 2015-09-05:
So far I had only been working with dxf exports, but looks like svg export already has this functionality. Nice.  Digging deeper into the dxf export, I see now that it is much more basic than the svg export.  It might make sense to bump up the dxf export functionality to include mapping models to polylines then using the model id for the layer name for the polyline overriding the path id.

danmarshall, 2015-09-07:
Hi @zrichmond thanks so much for the feedback, and even more thanks for being the first active participant / contributor. So glad that you find it as useful as I do and that it benefits you in a real world sense.
You are correct that the DXF export is more basic than the SVG. Simply because I don't "use" the DXF after I generate it, I just usually upload it to BigBlueSaw and let them cut it. So your input is invaluable as a DXF user. And since there are already hooks as you've seen in the SVG exporter it should be easy to add similar DXF structuring.
If you say you're green at JavaScript, the you might be even greener with TypeScript. However I'm excited to have contributors so let me know if you feel like tackling this or not. I would be happy to.

zrichmond, 2015-09-09:
Took a quick look at TypeScript, the .ts file extension makes sense now.  Seems like it is a way to make javascript somewhat typed, I could see it being very helpful particularly for developers coming from Java and C# or other strong OOP.   I could definitely see it being helpful for creating modules/libraries as you've done with maker.js.  

Anyhow, back to the dxf exporting.  Knowing that you are uploading dxfs to BigBlueSaw it makes sense that you wouldn't need layers mapping to specific tooling since it looks like they focus on water-jet and laser cutting neither of which, I believe, have different tools or cut types beyond center-line, with varying feed speed and intensity to get the desired cut quality for the given material.  My experience is with a CNC router/mill, particularly one with a tool changer, where there are many different types of tools available and each tool can make different types of cuts.  In order to facilitate this, typically there is a mapping of layers to the cut type, which would include specifics for the tool as well, radius, feed speed, spindle speed, cut depth, to name a few.

Looking at the dxf export, maybe a few steps could be:

Minimal:

 * Something simple like, if the object is an IModel, then export the model's paths to a layer name based on the model.id

More involved:

 * Since some router tool machining is based on the inside or outside of "closed" geometry, the addition of a "closed" property could also be helpful. This would indicate the geometry should be a closed path that would export as a POLYLINE, onto a layer name, again, based on the model.id

 * Add property "thickness" to export as dxf line thickness which in some 2.5D CAM packages, indicates depth of cut. Essentially a way to get a z-axis without full 3D CAD.

 * Colors for the layers, though it would be a bit more involved and not as elegant as SVG styling via CSS.

This is something I can personally take a closer look at, but I definitely wouldn't be disappointed if you write the code.  In fact just being able to see how you'd approach it might help me learn the library a bit more.

danmarshall, 2015-09-09:
Great, let's start with implementing the simple as a first pass. For the more involved feature, you mentioned "closed" geometry. I have actually been contemplating adding a feature I call "follow" which will attempt to make a continuous series of connected paths. Currently, there is no notion of connectedness at all (which had its advantages). 

One issue to solve is the fact that a continuous "followed" series might actually be made of paths from 2 separate models. For example, if you have 2 overlapping rectangle models, and do a combine operation to keep only their intersection, the result would be a new rectangle but it would have 2 sides from one model and 2 from another. But to your need of a tool path, this would need to be one polyline. So it looks like I ought to implement the "follow" function for that. 

That's fine, because I'd surmised that someone would rather also have a "polyline" equivalent in the SVG side as well.

zrichmond, 2015-09-09:
"Follow" is definitely a similar concept.  The main problem this addresses on a router is to control which side of the path the tool should cut.  In 2D cutting there are basically two ways to address the problem, either with direction and handing (left-hand or right-hand), or having "closed" geometry so you can be cutting either inside or outside using clock-wise or counter clock-wise to indicate direction.  It's important because one needs to be able to program offsets based on the shape of the tool, to account for the width of the tool path, kerf if it's a saw cut.  The direction is also important because the rotation of the tool effects how the chips of material are being removed as the tool is cutting.  In a dxf the direction is handled simply by order of the vertices in a polyline.  Also a polyline as defined in a dxf, doesn't actually have to be "closed" it's really just an ordered group of connected lines and arcs (arcs hidden in the bulge property of a vertex). Looking at my previous comment it looks like I miscommunicated that somewhat. With simple lines and arcs, dxf polyline and svg path are basically the same thing.  Maybe not exactly what you were hoping to get into with this, but hopefully it helps.

danmarshall, 2015-09-09:
Ok I've created an image for reference. Given 2 shapes, the blue plus and the green ring (which would be 2 models with their own paths) if they were to be combined as shown, you would want to have 5 "tool path layers", one for each of the four quadrants, and one for the outer circle, is this what you are needing?
[image]

zrichmond, 2015-09-12:
Yes I think that would work well, with the additional detail of being able to arbitrarily the name the paths so they can export as layer name (dxf) or group id (svg).  For example, being able to set the outer circle name, to 'outside', then each quadrant  to say 'pie slice', 'cake slice', 'pocket', inside_pocket_2'.  

I am sure you recognize the example if done the way you are suggesting would be pretty complicated. More likely in my experience the drawing of paths/geometry would be more straight forward. With 2D/2.5D machining I wouldn't typically be doing boolean type operations on basic shapes to build up more complex shapes.  Without being more familiar with your codebase,  I'm not sure which approach would be simpler.

danmarshall, 2015-09-18:
Yes the boolean operations do add some complexity. That will need some more thought. I'm going to proceed with the original plan of bringing the DXF export up to parity with the SVG export, as a start.

zrichmond, 2015-09-18:
Makes sense to me.  Thanks for taking a closer look at it and if there I can help I will.

danmarshall, 2015-09-26:
Hmmm at first glance it doesnt appear that DXF supports nested layers, i.e. a layer that contains another layer etc. In your other CAD software, are you able to create nested layers?

zrichmond, 2015-09-28:
No, DXF layers are simpler than the svg spec in that there isn't a way to nest them.   However, that being said,  DXF has a BLOCKS  section as a way to group entities. However, it's probably not worth using it to replicate svg's more sophisticated groups especially since the cam applications I've used usually don't know what to do with blocks.   

Maybe one way could be to have a layer/group property that once set by user code would be used for the layer/group name when exporting.  Could be as simple as having this property for both a path and a model, such that if it's set on the model the contained paths would also be set to the layer/group name.  Then if the export could it done based on the path, it could support both a model with various path names, and a model with single path names. The model name could be ignored since it's a container for the paths.  Pretty sure that would cover most of what would be needed for controlling the layer/group name.

danmarshall, 2015-09-28:
The layer/group property seems like a good idea. I wonder how this would also translate to the SVG side of things. Would be great to have one programming concept that works for any export type.

zrichmond, 2015-09-28:
Digging deeper into the svg export made me think that maybe it's as simple as just using the path's parent model id as the dxf layer name?

danmarshall, 2015-09-28:
That's what I started looking at - but then I hit nesting.

zrichmond, 2015-09-28:
Maybe ignore the nesting for dxf export?    In other words use just the immediate parent for the layer name, leaving deeper nesting a feature of svg output, and if someone is exclusively exporting for dxf they wouldn't gain from more complex nesting of models, but it is ultimately a limitation of the dxf format not the rest of the library.

danmarshall, 2015-09-30:
How do you feel about calling this "Z" instead of layer?

zrichmond, 2015-09-30:
I'd be a bit apprehensive to use Z.  In 2.5D machining, even though the cut path is generally 2D, a Z value may be used for the depth of cut, and sometimes thickness is used.   To confuse the issue further,  DXF format specifies allows entities to have XYZ coordinates and/or thickness.  I don't think svg has the concept of Z-order either, so maybe in leaning toward the flexibility of the svg format just calling it "group" would suffice?

danmarshall, 2015-09-30:
I understand your apprehension, it makes sense. Z is one possible interpretation of how someone might use the group. SVG does use CSS, so we could apply the concept to "z-index". That's why I suggested Z, so that both SVG and DXF have a use for this property. Since SVG will use z-index, I think your original suggestion of naming it "Layer" seems to work for both cases.

danmarshall, 2015-09-30:
Also, I was going to implement the Layer as a number, so that it can work as z-index for SVG, and Layer index in DXF. Were you hoping to use strings or names of layers, or will a number work?

zrichmond, 2015-09-30:
Layer names in a dxf are strings, and usually used in a human readable way so I guess I was thinking string for the type.  Also with the inclusion of css the z-index makes sense.  A layer in a cad drawing in the cad applications I've used at least, typically doesn't refer to the draw order so that might be confusing.  Might be that there are two properties layer: string and z-index: number ?  Maybe the answer is to have both?  That way it wouldn't be confusing if someone only had experience with web graphics vs cad formats.

danmarshall, 2015-09-30:
I use DraftSight and they have a default layer named 0, which led to the z-index thought. The implementation as numbers was pretty easy, just finished coding it. Was hoping to get one concept that worked for both. Bu I understand if using a number adds confusion. I suppose I could make it a string, and if SVG folks need it, then they can put a number in there...

danmarshall, 2015-09-30:
Ok, it finally made sense to use a string for layer, and to use SVG group to implement layer (not z-index). New version is 4.3, let me know how it works!

zrichmond, 2015-10-01:
Will do!

zrichmond, 2015-10-01:
I've only tried a few exports, but it's looking great. Nice work!

closed 2015-10-01 by zrichmond