microsoft/maker.js #86: Import SVG?

danmarshall, 2016-01-08:
Would like to know if anyone would find this feature useful.

NickNaso, 2016-01-10:
In my case it's useful because i work in a mixed team programmer and designer so when i found makerjs and i start to experiment i looked up in the api a function to load the svg. So i  can reuse work made by designer and in some case i could be more fast in my work.

danmarshall, 2016-02-13:
For this to happen, will depend on Bezier support. https://github.com/Microsoft/maker.js/issues/96

danmarshall, 2016-09-19:
Closing this issue for now. 
There's a lot of SVG (such as fills, gradients, etc.) that does not have a similar concept in Maker.js which only deals with outlines.
I've added a feature to import just SVG path data: http://microsoft.github.io/maker.js/docs/api/modules/makerjs.importer.html#fromsvgpathdata

closed 2016-09-19 by danmarshall