microsoft/maker.js #87: Unit tests

danmarshall, 2016-01-08:
Need these! Can't believe we got this far without em.

danmarshall, 2017-07-18:
@xuyuji9000 thank you, have you seen the current tests in https://github.com/Microsoft/maker.js/tree/master/test

I am using Mocha.

xuyuji9000, 2017-07-19:
As far as I'm understanding this /test:

 - manual is for environment test (browser environment, node environment, not understand worker)

> first time using typescript, going through the typescript docs, before I can understand more about test/manual

- Beziers, Core, Paths  are three core starting points of the test suite.

danmarshall, 2017-07-19:
A Web Worker is a JavaScript background thread, in the browser. Since it is background, it does not have access to UI like the DOM or window objects. 
See https://developer.mozilla.org/en-US/docs/Web/API/Web_Workers_API

Maker.js will work in Node, Browset UI thread, or this Browser background thread (Web Worker). It needs to know about its environment so that it can decide how to load a dependency.

So it is in the /manual folder because I haven't automated it. Mocha runs in Node, thus it wouldn't test the other environment types.

Mocha will run every file in the /test folder. So yes, those are the only tests I have so far.

xuyuji9000, 2017-07-23:
Got this error,  I ran `npm run build`

> src/core/openjscad.ts(133,21): error TS2394: Overload signature is not compatible with function implementation

After I commented the error line 133, rerun `npm run build`

> playground/worker/export-worker.ts(133,13): error TS2322: Type '{ (modelToExport: IModel, options?: IOpenJsCadOptions): string; (pathToExport: IPath, options?: I...' is not assignable to type 'IExporter'.
  Types of parameters 'options' and 'options' are incompatible.
    Type 'IExportOptions' has no properties in common with type 'IOpenJsCadOptions'.
playground/worker/export-worker.ts(151,13): error TS2322: Type '(model: IModel, options: IOpenJsCadOptions) => string' is not assignable to type 'IExporter'.
  Types of parameters 'options' and 'options' are incompatible.
    Type 'IExportOptions' has no properties in common with type 'IOpenJsCadOptions'.

Stuck on this right now.

danmarshall, 2017-07-23:
Hi @xuyuji9000 , thanks for reporting. The new TypeScript 2.4 is very impressive! It is catching some good bugs. I've made the changes to fix this, can you please pull and try again?

closed 2017-12-27 by danmarshall