microsoft/maker.js #90: Layers in DXF export

NickNaso, 2016-01-08:
Hi, we just discovered this cool library.
Great stuff!
We have a request for you. How we can manage or obtain layers in exporting DXF?
In svg it worked in this way:

```javascript
var fs = require('fs');
var makerjs = require('makerjs');
//some code to build line and circle
var cut = makerjs.exporter.toSVG('', {svgAttrs: {id:'Cut'}});
var decoration = makerjs.exporter.toSVG('', {svgAttrs: {id:'Decoration'}});
var svgToexport = cut + decoration;
fs.writeFileSync('test.svg', svgToexport);
```

I there any solution for handle this for the dxf file?

Layers are very important for machine to handle better different elements such as in this case decoration and cut.

All the best.

carvadero, 2016-01-08:
Thank you Nick! :)

Does this work for you?:

```javascript
var makerjs = require('makerjs');

//some code to build line and circle
var myDrawing = {
  myCircle: new makerjs.paths.Circle([0, 0], 3),
  myLine: new makerjs.paths.Line([0, 3], [3, 3])
};

myDrawing.units = makerjs.units.Inch;
myDrawing.myCircle.layer = 'myLayer1';
myDrawing.myLine.layer = 'myLayer2';

console.log(makerjs.exporter.toDXF(myDrawing));
```

NickNaso, 2016-01-09:
@carvadero Thanks now i have layer in my dxf file on monday i test the file on my machine print and say if it is all ok!

danmarshall, 2016-01-09:
Also see related [Issue 57](https://github.com/Microsoft/maker.js/issues/57) - this feature was requested by @zrichmond and will hopefully work for you too :)

danmarshall, 2016-02-01:
I fixed a small bug which may have prevented output of layers on paths. Can you try it now @NickNaso ?

NickNaso, 2016-02-03:
@danmarshall Just tested and it seems to be ok. I removed my previous version of makerjs 0.6.7 and updated to 0.6.10 and tested it.

danmarshall, 2016-02-05:
ok great, I'm going to close this issue then.

closed 2016-02-05 by danmarshall