microsoft/maker.js #145: selectable models

themaniac, 2016-04-28:
Is it possible to select areas like circles or squares for programmatically adding other nested models inside?

danmarshall, 2016-04-28:
Do you mean like a clipping path? If so, yes. Using the [combine function](http://microsoft.github.io/maker.js/docs/api/modules/makerjs.model.html#combine).

Combine needs 2 models, so even a simple circle path will need to be wrapped in a model.

Here is an example of clipping the logo inside of a circle. Copy and paste into the [playground](http://microsoft.github.io/maker.js/playground/)

```
var makerjs = require('makerjs');
var logo = require('makerjs-logo');

function test() {
  
  this.models = {
    logo: new logo(),
    clip: {
      paths: {
        c: new makerjs.paths.Circle([0,0], 9)
      }
    }
  };

  makerjs.model.combine(this.models.logo, this.models.clip, true, false, true, false);
  
}

module.exports = test;
```

themaniac, 2016-04-28:
hi dan, i mean:

i create a square with 2 circles inside, then i want to select (clicking on it for example) one of the circles, then do something with js code (adding another model in it)...

tell me if it's possible.

Thanks in advance

danmarshall, 2016-04-28:
I see. Yes this can be accomplished with a technique called routes. A route is simply an array of property names to get a deep reference to an object.
For example, with this example object:
```
{models:{foo:{models:{bar:paths:{abc:...}}}}}
```
you can reference abc with a route like ['models','foo',models','bar','paths','abc']

This technique is used in the playground, when you click the checkbox named "show path names" you can see a demo of it in action. When you click the text of a path name, the path will become selected and you can see its route.

The output SVG is a reflection of your model. Models become the SVG < g > tag which can be nested. The route can be deconstructed by a click handler on the SVG. Here are links to where these happen:
* [Click Handler](https://github.com/Microsoft/maker.js/blob/master/playground/playground.ts#L458)
* [Crawl SVG elements to build a route](https://github.com/Microsoft/maker.js/blob/master/playground/playground.ts#L324)
* [Using a route to reference the model](https://github.com/Microsoft/maker.js/blob/master/playground/playground.ts#L385)

If you think this will be a useful feature, perhaps we can build this into the output SVG.

themaniac, 2016-05-06:
Yes, it could be a useful feature, thank you!

danmarshall, 2017-05-25:
There is now a `data-route` attribute on elements which is a stringified array of the route to the element. It is only available in [separate mode](http://microsoft.github.io/maker.js/docs/exporting/#SVG) of SVG.

closed 2017-05-25 by danmarshall

z3dev, 2017-05-26:
Wouldn't this then be FABRIC?

danmarshall, 2017-05-26:
@z3dev - I'm not familiar with FABRIC - can you provide a link?

z3dev, 2017-05-26:
http://fabricjs.com

It's awesome by the way.

danmarshall, 2017-05-27:
@z3dev oh yes now I recall I've seen it - very cool!
This little feature just lets you re-connect the SVG with your object model. Someone would need to build a lot more to compare to Fabric :)