microsoft/maker.js #146: AngularJs Integration

themaniac, 2016-04-28:
Hi all,

i'm trying to integrate this beautiful tool into an angularjs project, but when i include makerjs's src core into the project i have a lot of error

`<script src="makerjs/src/core/maker.js"></script>
    <script src="makerjs/src/core/angle.js"></script>
    <script src="makerjs/src/core/point.js"></script>
    <script src="makerjs/src/core/units.js"></script>
    <script src="makerjs/src/core/equal.js"></script>
    <script src="makerjs/src/core/measure.js"></script>
    <script src="makerjs/src/core/path.js"></script>
    <script src="makerjs/src/core/break.js"></script>
    <script src="makerjs/src/core/paths.js"></script>
    <script src="makerjs/src/core/model.js"></script>
    <script src="makerjs/src/core/combine.js"></script>
    <script src="makerjs/src/core/collect.js"></script>
    <script src="makerjs/src/core/simplify.js"></script>
    <script src="makerjs/src/core/loops.js"></script>
    <script src="makerjs/src/core/expand.js"></script>
    <script src="makerjs/src/core/kit.js"></script>
    <script src="makerjs/src/core/exporter.js"></script>
    <script src="makerjs/src/core/dxf.js"></script>
    <script src="makerjs/src/core/xml.js"></script>
    <script src="makerjs/src/core/svg.js"></script>
    <script src="makerjs/src/core/openjscad.js"></script>
    <script src="makerjs/src/core/solvers.js"></script>
    <script src="makerjs/src/core/fillet.js"></script>
    <script src="makerjs/src/core/intersect.js"></script>
    <script src="makerjs/src/models/BoltCircle.js"></script>
    <script src="makerjs/src/models/BoltRectangle.js"></script>
    <script src="makerjs/src/models/Dome.js"></script>
    <script src="makerjs/src/models/RoundRectangle.js"></script>
    <script src="makerjs/src/models/Oval.js"></script>
    <script src="makerjs/src/models/OvalArc.js"></script>
    <script src="makerjs/src/models/ConnectTheDots.js"></script>
    <script src="makerjs/src/models/Rectangle.js"></script>
    <script src="makerjs/src/models/Ring.js"></script>
    <script src="makerjs/src/models/Polygon.js"></script>
    <script src="makerjs/src/models/Square.js"></script>
    <script src="makerjs/src/models/Slot.js"></script>
    <script src="makerjs/src/models/SCurve.js"></script>
    <script src="makerjs/src/models/Star.js"></script>`

below the errors

`Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: unitType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: unitType is not defined
Uncaught ReferenceError: unitType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught ReferenceError: pathType is not defined
Uncaught TypeError: Cannot read property 'easing' of undefined
Uncaught TypeError: Cannot read property 'offsetLeft' of null
`

please help me, i'm stuck and i want to use makerjs so hard!

danmarshall, 2016-04-28:
Which version of Angular- 1 or 2?
Looks like you are trying to include each source file, I'm not sure that the module system is Angular-friendly since some of the modules are spread across files. Can you reference the pre-built version?
http://microsoft.github.io/maker.js/target/js/browser.maker.js

themaniac, 2016-04-28:
angular 1, i want, if it's possible, to clone the playground style on an angularjs project and start from it. You know something similar? any documentation or plunkr example?

thanks

danmarshall, 2016-04-29:
Ok, here is a simple plunkr example using Angular 1.5 https://plnkr.co/edit/NCr7Bf?p=preview

themaniac, 2016-04-29:
Can i use the playground grid aswell?

danmarshall, 2016-04-29:
The background is actually just a simple static jpeg with a grid image. The grid in the image does not relate to the units of your model. So go ahead and just use CSS and an image in your Angular app. 
In the future, having a dynamic grid with unit spacing would be a cool new feature.

themaniac, 2016-04-29:
I don't understand... on the demo playground i can select, move etc...

how it's possible with a static jpeg/grid?

thanks for your support

themaniac, 2016-04-29:
Oh, now i see you load other libraries on playground, like pep.js and stuff like this... what's are usefull to me to clone playground?

thanks

danmarshall, 2016-04-29:
It depends on the functionality you want in your Angular app. If you want all functionality, then take all scripts :)

* pep.js, pointer.js - multitouch support
* marked.js - Markdown support
* codemirror.js - JavaScript editor
* export-format - Maker.js export formats

danmarshall, 2016-04-29:
Also, you can just fork this repo, and modify your own copy in your fork. The gh-pages branch will automatically become a website for you, just as you see here on GitHub.

themaniac, 2016-05-04:
thanks for the reply, i need to know if i could contact you in private.

Thanks in advance

danmarshall, 2016-05-06:
Hi Gianluca @themaniac ,  sure you can email makerjs@microsoft.com

themaniac, 2016-05-10:
@danmarshall i forked the repository and i'm trying to use :

> pep.js, pointer.js - multitouch support

i don't know how to implement it to angularjs, i've only the svg static into the grid... please help me!

danmarshall, 2016-05-10:
@themaniac I don't see your fork in the [list of forks](https://github.com/Microsoft/maker.js/network/members) nor in [your repos](https://github.com/themaniac?tab=repositories) - can you post a link? Also, I may be able to help you offline, email makerjs@microsoft.com

themaniac, 2016-05-10:
I've mailed you a couple of days ago on that address :)

But if i cannot import svg or dxf into makerjs i'm stuck on my project :(

closed 2016-05-19 by danmarshall