microsoft/maker.js #150: DXF manipulation

themaniac, 2016-05-05:
Hello, is it possible to :

1- load a DXF file;
2- parse to SVG in order to view it;
3- edit through makerjs;
4- re-export to DXF; 

i know it's possible from step 2 to 4... didn't found a solution for the first step above. It's important because i have to start from existing dxf files...

Please let me know and thanks again

danmarshall, 2016-05-06:
Hi Gianluca, there is currently no import, but this could be a nice feature. Would like to see if others need this also.

danmarshall, 2016-09-20:
Looks like there are a few DXF parsers on NPM:
* https://www.npmjs.com/package/dxf
* https://www.npmjs.com/package/dxf-parsing
* https://www.npmjs.com/package/dxf-parser
* https://www.npmjs.com/package/node-dxf

I am guessing there's an easy mapping to Maker.js models.

closed 2016-09-20 by danmarshall