microsoft/maker.js #155: SVG import and how to add id to svg object

themaniac, 2016-05-10:
I saw in another question that's possible to import SVG into maker.js ... where can i find the snippet in the repository?

Then, using this code (angularjs project) : 

` var makerjs = require('makerjs');

    var model = {
        paths: {
            head: new makerjs.paths.Circle([0, 0], 90),
            eye: new makerjs.paths.Circle([25, 25], 10),
            mouth: new makerjs.paths.Arc([0, 0], 50, 225, 315),
            wink: new makerjs.paths.Line([-35, 20], [-15, 20])
        }
    };

    var svg = makerjs.exporter.toSVG(model);

    $scope.myHTML = $sce.trustAsHtml(svg);`

How can i add an ID as an attribute to this exported SVG? 

thanks in advance

themaniac, 2016-05-10:
resolved id attrs like this : 

`var svg = makerjs.exporter.toSVG(model, {svgAttrs: {id: 'drawing'}});`

waiting for help on the svg import ... :)

danmarshall, 2016-05-10:
SVG import is still only a proposal. And it has a dependency on Bezier support, so it might be a while.
https://github.com/Microsoft/maker.js/issues/86

closed 2016-05-11 by danmarshall