microsoft/maker.js #168: Note. It may be worth noting in your examples that models and paths can be arrays

rogerclarkmelbourne, 2016-07-24:
Hi Dan

I'm not sure if this is intentional (I think its just the way js works ;-) )

But its worth noting that you don't need to name the paths or the models, both paths and models work OK as arrays.

As an example I modified your origin example to push a load of boxes into an array

https://jsfiddle.net/ddkqk1s8/

I'm using this technique as I need to create hundreds of cutting lines to make plywood bendable , when laser cut, and its not practical to name every line (though I suppose I could programmatically name them e.g. line0, line1,line2 etc etc)

danmarshall, 2016-07-24:
Hi Roger,
I believe you are right - it's just the way js works. 
This is brilliantly simple! I do appreciate you pointing this out. I'm going to leave this issue open for visibility.
Since I'm using TypeScript, there is a strictness that keys can only be a string. But your trick is great to know for straight js developers.

rogerclarkmelbourne, 2016-07-24:
No worries

You could add this to the documentation / examples. (I presume you have access to the github.io private files ?)

danmarshall, 2016-07-27:
All the docs are also open source - take a look at the gh-pages branch. https://github.com/Microsoft/maker.js/tree/gh-pages/docs

danmarshall, 2016-08-13:
Hi Roger, I just realized came across a scenario where this will create an undesirable result. In a boolean operation, sometimes Maker.js needs to break a path into pieces, then add the pieces back into the model. To do this, it creates a [similar path id](http://microsoft.github.io/maker.js/docs/api/modules/makerjs.model.html#getsimilarpathid) which will not be numeric. So when you go to inspect your array, you may not see these pieces.

closed 2016-08-13 by danmarshall

rogerclarkmelbourne, 2016-08-14:
Dan
Thanks for the info.

I have not seen any problems so far, with creating "living hinges" out of laser cut wood, which need lots (hundreds) or small isolated lines.

And I'm also creating things like dovetail joints, which are effectively polylines - but I'm not doing any manipulation of these objects after I create them, apart from scaling and rotating so that they can be displayed differently from being exported to DXF.

Cheers

Roger

danmarshall, 2016-08-14:
Hi Roger, on a related note, if possible can you post a photo of your project? I love seeing the stuff people are making with Maker.js :)

rogerclarkmelbourne, 2016-08-14:
Hi Dan

Its a bit of a work in progress, but I'll try to take some photos. 

I've been making laser cut bangles out of 3mm plywood, which work fairly well, except I can't get them to clip together just using the wood on its own. I'd tried using dovetail joints etc, but I need to buy some commercial clasps to hold the ends together.

I've been experimenting with different distance between the cuts along the length of the bangle (using Math.sin) to achieve a non round final bend profile, but its trick to get the parameters correct.

I've also been experimenting with possibly using bangle profile (shape) data by defining the shape in Inkscape, and then exporting to SVG, which I then parse and attempt to calculate the distance between each cut to get the overall profile as defined in Inkscape - but this is even harder, so I've put that on the back burner until I have more time ;-)