microsoft/maker.js #182: Placing multiple texts in a drawing

lazyoldjack, 2016-08-13:
I've probably missed something obvious and/or am not using maker.js properly but the only way I seem to be able to place some text anywhere other than at 0,0 is to run through and move the origin of every part of the generated text. Would be nice to just give it an origin when requesting the text.

Also, without some more jiggery-pokery, subsequent texts overwrite previous texts as I add them to my overall drawing model. But like I say, I probably just don't see how I'm supposed to do these things the right way.

danmarshall, 2016-08-13:
HI @lazyoldjack, Text is a model just like any other, so you will just set the `origin` property of your Text model. All of the child characters will be relative to this parent.

Fonts do have an interesting behavior, in that the first character is not necessarily referenced at x=0. That's up to the font author where to draw their shape.

I've added a new function `model.zero` (which I have yet to document) which moves any model's bottom left renderable path to [0,0]. Bear in mind that this does not always move that model's origin to [0,0] but to some offset that will make the renderable drawing begin at [0,0]. From this you can just `model.moveRelative' to a known point.

This example begins with the same code from http://microsoft.github.io/maker.js/playground/?script=Text
```
var makerjs = require('makerjs');

function Text(fontPath, text, fontSize, combine) {

    var _this = this;

    //load a font asynchronously
    opentype.load('/maker.js/fonts/' + fontPath, function (err, font) {
        if (!err) {

            var textModel = new makerjs.models.Text(font, text, fontSize, combine);
          
            _this.models = textModel.models;

            //To move text: bring the bottom left corner of actual rendering to [0, 0]
            makerjs.model.zero(_this);
            //Now move it to the place you want it
            makerjs.model.moveRelative(_this, [0, 20]);

           //You also could simply do this, but every font will probably set somewhere slightly off
           //_this.origin = [0, 20];

            //"playgroundRender" is strictly a Playground function - in your app, call your own callback.
            playgroundRender(_this);
        }
    });

}

Text.metaParameters = [
    {
        title: "font", type: "select", value: [
            'allertastencil/AllertaStencil-Regular.ttf',
            'blackopsone/BlackOpsOne-Regular.ttf',
            'emblemaone/EmblemaOne-Regular.ttf',
            'keaniaone/KeaniaOne-Regular.ttf',
            'montserratsubrayada/MontserratSubrayada-Regular.ttf',
            'plaster/Plaster-Regular.ttf',
            'sirinstencil/SirinStencil-Regular.ttf',
            'stardosstencil/StardosStencil-Bold.ttf',
            'stardosstencil/StardosStencil-Regular.ttf',
            'wallpoet/Wallpoet-Regular.ttf'
        ]
    },
    { title: "text", type: "text", value: 'Hello' },
    { title: "font size", type: "range", min: 10, max: 200, value: 72 },
    { title: "combine", type: "bool", value: false }
];

module.exports = Text;
```

danmarshall, 2016-08-13:
Perhaps this is a simpler example:

```
var makerjs = require('makerjs');

//load a font asynchronously
opentype.load('/maker.js/fonts/allertastencil/AllertaStencil-Regular.ttf', function (err, font) {
  if (!err) {

    var words = {
      models: {
        hello: new makerjs.models.Text(font, 'Hello', 100),
        world: new makerjs.models.Text(font, 'World', 87)
      }
    };
    
    words.models.hello.origin = [0, 110];
    words.models.world.origin = [0, 20];

    playgroundRender(words);
  }
});
```

In the Playground, sometimes you need to click RUN twice to download the font first.

lazyoldjack, 2016-08-15:
Thanks for the clear explanation Dan; hopefully this will help other noobs too! :o)

closed 2016-08-15 by lazyoldjack