microsoft/maker.js #185: Color?

lazyoldjack, 2016-08-16:
Whilst debugging a relatively complex drawing generator I find myself thinking that it would be useful to be able to control the display color of models/paths. Though color probably would never play a part in a final drawing for cut, I would be able to use it to spot where this particular line is in the drawing, because currently it's hiding over another black line...

danmarshall, 2016-08-17:
Hi John, check this JsFiddle: https://jsfiddle.net/danmarshall/5u7Lr02n/
You're moving faster than I'm able to document :) but that's a good thing. Take a look at this document in progress: https://github.com/Microsoft/maker.js/blob/gh-pages/_snippets/svg-styling.html

The answer is you can use standard CSS styling, and that you should pass an options object to ```exporter.toSVG``` to enable the non-continuous SVG mode:

```
var svg = makerjs.exporter.toSVG(model, { useSvgPathOnly: false });
```

In continuous mode, the export follows paths in a chain, and outputs the chain as one SVG path element. In non-continuous mode, each path is a single SVG element, and models become an SVG <g> group element, with all IDs being the same as your model.

lazyoldjack, 2016-08-18:
Ah, this looks excellent - thanks!

closed 2017-01-03 by danmarshall