microsoft/maker.js #189: Best material usage

lazyoldjack, 2016-08-22:
'Twould be nice to pass a collection of models to a routine, along with the material size, for it to return the collection optimally placed for cutting. Of course, it would also be nice to win the lottery :o)

Anyway, I mention it in case anyone out there likes a (big) challenge; I fear it's a bit outside my capability.

danmarshall, 2016-08-22:
Check out this fantastic project: https://github.com/Jack000/SVGnest

lazyoldjack, 2016-08-22:
Wow, that is some serious software. Just what I need I think. Will 
investigate in detail later! Thanks Dan, you seem to have everything 
covered, yet again!

Cheers

John

------------------------------------------------------------------------

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2016-09-10 by danmarshall