microsoft/maker.js #192: Drawing text paths

davejack1, 2016-08-29:
I am building a model by pushing paths into an array. If I am not mistaken makerjs does not allow to push models into an array, am I correct?

Now I came to a point that I have to add text to my model and as far as I checked the example on documentation page, text object is defined as model. So, my first question is; Is it possible to convert this model into a path and then push it into a model. I will be creating multiple text models within a for loop, hence the need of pushing into an array.

All those aside my first attemp for creating a text failed;

```
var makerjs = require('makerjs')
var textObj = new makerjs.models.Text("verdana", 'Hello', 12, false);
var svg = makerjs.exporter.toSVG(textObj)
document.write(svg)
```

Where am I wrong?

danmarshall, 2016-08-29:
Hi @davejack1, I think I see a couple things going on:
1. Models may contain other models. Here is a small example:
```
var myModel = {
  models: {
    "a": modelA,
    "b": modelB
  }
}
```
This is a tree structure, so it isn't flat like an array. You can give each item within ```myModel.models``` an id. The id can be a number. Here is an example [from the documentation](http://microsoft.github.io/maker.js/docs/intermediate-drawing/#Cloning)
```
var wheel = { models: {} };
var numTeeth = 30;
var tooth = new sawtooth(numTeeth, 100, 20, 10);

for (var i = 0; i < numTeeth; i++ ) {
    var clone = makerjs.cloneObject(tooth);
    var a = 360 / numTeeth;
    makerjs.model.rotate(clone, a * i, [0, 0]);
    wheel.models[i] = clone;         //use i as the id
}
```

2. You will need to use a font object for ```makerjs.models.Text``` - "verdana" is a string. Check out https://github.com/nodebox/opentype.js for details on how to load a font from a URL. Also see http://microsoft.github.io/maker.js/docs/advanced-drawing/#Fonts%20and%20text

davejack1, 2016-08-29:
Thank you for your answer Dan.

Shortly after I posted, realized that the only way is loading an external font and did so. However, I am having a problem with the global variable which has been assigned the text model.

Below my attempt;

```
var makerjs = require('makerjs')
var textModel;

opentype.load('fonts/Roboto-Black.ttf', function(err, font) {
    if (err)
    {
        alert('Could not load font: ' + err);
    }
    else
    {
        textModel = new makerjs.models.Text(font, 'Hello', 72, false);
    }
});

var svg = makerjs.exporter.toSVG(textModel);
document.write(svg);
```

It does not work. However, if I put the 

```
var svg = makerjs.exporter.toSVG(textModel);
document.write(svg);
```
lines into the `else` statement, it actually does work, but I need to be able to reach the value assigned `textModel` variable outside the callback function.

danmarshall, 2016-08-29:
You are now using an asynchronous function (waiting for a download) so there are ramifications to the way your app is coded. You may try ```document.getElementById('yourDivId').innerHTML = svg;``` instead of ```document.write(svg)``` inside of the callback function.

danmarshall, 2016-08-29:
Moving a model can be as simple as setting its origin property:
```myModel.origin = [100, 0];```
See http://microsoft.github.io/maker.js/docs/basic-drawing/#Moving

Also with Text I also find zeroing and centering useful: http://microsoft.github.io/maker.js/docs/intermediate-drawing/#Zeroing and Centering

davejack1, 2016-08-29:
Thanks Dan, Zeroing and centering might come in handy!

danmarshall, 2016-08-29:
Also, I'm considering of adding boolean params to centering ```(centerX, centerY)``` so that you can center one dimension or the other.

davejack1, 2016-08-31:
Hey Dan,

I am having trouble with `center` function for the text model

Below is my example (Again I could't make it run but...)
https://jsfiddle.net/0x2Lhbep/

danmarshall, 2016-09-01:
Hi @davejack1 , I took a look at your JsFiddle. I think that JsFiddle is not downloading scripts right now as they are trying to fight spam.

Anyways I got your example to work on a local web site after after fixing:
* url to the font. Looks like you cannot load fonts cross domain. So I downloaded the font manually and put it in my local web site.
* remove line of code that "uncentered" :)
```
makerjs.model.center(finalModel.models[0]);
//finalModel.models[0].origin = [0, 0];
```

closed 2016-09-10 by danmarshall