microsoft/maker.js #194: Combining paths and models

davejack1, 2016-08-30:
I had asked this as a side question in my [previous post](https://github.com/Microsoft/maker.js/issues/192). However, it did not solve my problem.

What I need to do is combining both paths and models in one variable and return it from a function.

Below I made a quick example;

I actually have several problems;

1. Not being able to put paths and models into one variable

2. Not being able to return the text objects generated within the callback function

```
var makerjs = require('makerjs')

function draw(rad)
{
    var finalModel = { models: {} };

    pathArray = [];
    pathArray.push (new makerjs.paths.Circle([0,0],rad)); 
    pathArray.push (new makerjs.paths.Circle([0,0],rad*2));

    finalModel = {paths: pathArray}

    opentype.load('fonts/Roboto-Black.ttf', function(err, font) {
        if (err)
        {
            alert('Could not load font: ' + err);
        }
        else
        {
            for (var i = 0; i<2; i++)
            {
                finalModel.models[i] = new makerjs.models.Text(font, 'Hello', 100*i, false);
                finalModel.models[i] = new makerjs.model.rotate(finalModel.models[i],50,[0,0]);
                finalModel.models[i].origin = [100,200*i];
            }
        }
    });

    return finalModel
}

document.getElementById('div1').innerHTML = makerjs.exporter.toSVG(draw(50));
```

This code draws the two circles but no Text is drawn. I also got the following error

`TypeError: undefined is not an object (evaluating 'finalModel.models[i] = new makerjs.models.Text(font, 'Hello', 100*i, false)')`

I suspect callback function of opentype leads some problems.

Being able to convert models into paths would be very useful for pushing them into an array. Or is there any alternative way to that?

Would really appreciate if you can help me on tackling this problem.

danmarshall, 2016-08-30:
Hi @davejack1 , can you create a [JSFiddle](https://jsfiddle.net/) with this example?

davejack1, 2016-08-30:
Here you are;
https://jsfiddle.net/skemstu4/
But I couldn't run into for some reason.

danmarshall, 2016-08-30:
Ok I see a couple of issues. We will look at them one at a time. Let's scope to just this section of code:

```
    var finalModel = { models: {} };

    pathArray = [];
    pathArray.push (new makerjs.paths.Circle([0,0],rad)); 
    pathArray.push (new makerjs.paths.Circle([0,0],rad*2));

    finalModel = {paths: pathArray}
```
The ```finalModel``` variable is being overwritten. At first, its value was ```{ models: {} }``` but at the end it was overwritten to ```{paths: pathArray}```. To add a property you should:
```
    finalModel.paths = pathArray;
```
Now, about that pathArray. Let's use an object and not an array. Maker.js will expect your ```paths``` property to be an object so that it can manipulate the collection of path objects and reference them by ID.

We can initialize everything in one line:
```
    var finalModel = { models: {}, paths: {} };
```
I'll encourage you to give your paths meaningful IDs like this:
```
    finalModel.paths.smallCircle = new makerjs.paths.Circle([0,0],rad); 
    finalModel.paths.largeCircle = new makerjs.paths.Circle([0,0],rad*2);
```
But you can just use a number like this:
```
    finalModel.paths[0] = new makerjs.paths.Circle([0,0],rad); 
    finalModel.paths[1] = new makerjs.paths.Circle([0,0],rad*2);
```
Or an iterator like this: (most similar to your ```Array.push```:
```
    var i =0;
    finalModel.paths[i++] = new makerjs.paths.Circle([0,0],rad); 
    finalModel.paths[i++] = new makerjs.paths.Circle([0,0],rad*2);
```

danmarshall, 2016-08-30:
Also, you will need to put this inside of the font load handler:
```
document.getElementById('div1').innerHTML = makerjs.exporter.toSVG(draw(50));
```

closed 2016-09-10 by danmarshall