microsoft/maker.js #198: Converting SVG to Canvas element

davejack1, 2016-09-19:
Is it possible to convert output of the `makerjs.exporter.toSVG()` method to a canvas element for faster drawing? Most of the existing examples either gets the SVG file as an URL or a text.

danmarshall, 2016-09-19:
Hi @davejack1 - Maker.js does not natively implement canvas since CNC machines usually need a vector format. But there are other projects that convert SVG to Canvas, take a look at https://github.com/canvg/canvg

danmarshall, 2016-09-21:
I see you mentioned faster drawing - so you are seeing slow rendering time of SVG? It's possible that in some cases that Canvas may be faster, but not always. How many elements are in your drawing?

davejack1, 2016-09-22:
I am changing the parameters of the SVG model with dat.GUI library and refreshing the DIV content with the new SVG output at the same time. Its response time is a bit slow, not instant. Tried canvas and it was not any better.

In my model there are a lot of text elements. I suspect that's the reason, since text drawing is async.

danmarshall, 2016-09-22:
To get better performance you can try a few things, depending on your scenario. If there are parts of your drawing that do not change in between GUI changes, you may want to cache a model object and not recompute it.

For SVG, you also might want to try to export in [separate mode instead of path-only mode](http://microsoft.github.io/maker.js/docs/exporting/#SVG) which will not do the computation of continuous chains. e.g. ```makerjs.exporter.toSVG(model, {useSvgPathOnly: false})```

closed 2016-09-30 by danmarshall