microsoft/maker.js #201: Curved text

davejack1, 2016-09-22:
Is it possible to create a curved text with MakerJS?
If not, is it possible to create a text SVG with another library and then combine it with the makerJS SVG variable? I need to convert the final result to DXF eventually

danmarshall, 2016-09-22:
Is your curve a circular arc or a Bezier curve? The circular arc would be pretty straightforward. The Bezier curve will involve a little more math... Ultimately I am planning on being able to attach text to any chain, but that may not be implemented very soon.

If you create SVG in another library then you can import it with [makerjs.importer.fromSVGPathData](http://microsoft.github.io/maker.js/docs/api/modules/makerjs.importer.html#fromsvgpathdata) which will convert it into a model.

davejack1, 2016-09-22:
I only need a circula arc.

davejack1, 2016-09-24:
[Here](http://jsfiddle.net/Qh9X5/9043/) is an example I attempted. (For some reason it does not show the circle I draw, but it is there)

Used the example code of the D3 library to create a curved text and then added it to the circle path I already created, but when I get the dxf output, I only see the circle, neither the text nor the curved path is there.

danmarshall, 2016-09-24:
Hi @davejack1, Maker.js does not currently import all types of SVG objects. The fromSVGPathData method only imports the "d" (data) part of an SVG path. So in your example, the only part that fromSVGPathData can accept would be "M 10,90 Q 100,15 200,70 Q 340,140 400,30". I'll work on an example of text on a circular arc.

davejack1, 2016-09-24:
I see. Would really appreciate it. Thank you.

danmarshall, 2016-09-26:
Hey @davejack1 - here's an example: http://microsoft.github.io/maker.js/playground/?script=arc-text

davejack1, 2016-09-26:
Thanks a lot. Will look into it in detail.

davejack1, 2016-09-28:
I have one quick question;

Although you get the point on the arc for the given angle and set this point as the origin of the character, the whole text sits on top of the arc.

`var p = makerjs.point.fromAngleOnCircle(angleFromEnd, arc);`
`char.origin = p;`

How can I put the text exactly on the arc?

danmarshall, 2016-09-28:
Isn't this the same behavior as SVG? The bottom of the character is on the path, as in this example http://apike.ca/prog_svg_path_text.html

davejack1, 2016-09-28:
[This](http://visihow.com/images/8/87/Write_text_on_a_circular_path_on_GIMP.mp4-canvas181_349997.jpg) is what I am trying to do. Sorry for the bad image quality, but this is the only example I could find.

What _actually_ I am trying to do is [this](https://www.signwarehouse.com/blog-support/wp-content/uploads/2015/02/Arc-Text_pic4.jpg). I tried to draw the lower part of this example. To do so, I simply rotated a half circle by 180 degrees. That way I don't get [this](https://www.signwarehouse.com/blog-support/wp-content/uploads/2015/02/Arc-Text_pic4.jpg) effect because non-rotated arc holds the text above it, whereas a rotated arc holds it inside the arc, just like [this](http://ltunlimited.typepad.com/.a/6a00e5521b79d888340120a8b8192d970b-pi) example (imagine its a full circle)

Putting the text exactly on the arc would solve this problem, or being able to put the text under the arc in your very first example would also yield a solution. However, I don't know how can I get the position of the top point of a character.

danmarshall, 2016-09-29:
So I think you are asking that the origin should be in the vertical center of the character. I've updated the sample to reverse the direction when it's on the bottom:
 * [on top example](http://microsoft.github.io/maker.js/playground/?script=arc-text#params=["allertastencil-regular","Hello world",72,1,200,20,160,true,true])
 * [on bottom example](http://microsoft.github.io/maker.js/playground/?script=arc-text#params=["allertastencil-regular","Hello world",72,1,200,200,-20,false,true])

Then you can try this code to bring it all together:
```
var makerjs = require('makerjs');
var ArcText= require('arc-text');

function EncircleText(font, topText, bottomText, fontSize, fontScale, arcRadius, topAngle, bottomAngle, showCircle) {

 
  	this.models = {
      topText: new ArcText(font, topText, fontSize, fontScale, arcRadius, topAngle, 180 - topAngle, true, showCircle),
      bottomText: new ArcText(font, bottomText, fontSize, fontScale, arcRadius, 180 + bottomAngle, 360 - bottomAngle, false, showCircle)
    }
  
  
}

EncircleText.metaParameters = [
    { title: "font", type: "font", value: '#stencil' },
    { title: "top text", type: "text", value: 'this is the top' },
    { title: "bottom text", type: "text", value: 'this is the bottom' },
    { title: "font size", type: "range", min: 10, max: 200, value: 72 },
    { title: "font scale", type: "range", min: .1, max: 2, step: .1, value: 1 },
    { title: "arc radius", type: "range", min: 1, max: 1000, value: 200 },
    { title: "top angle", type: "range", min: 5, max: 45, value: 15 },
    { title: "bottom angle", type: "range", min: 5, max: 45, value: 15 },
    { title: "show circle", type: "bool", value: true }
];

module.exports = EncircleText;
```
(this only will work in the playground because "arc-text" isn't actually published on NPM)

You can see that when the bottom text 'this is the bottom' is not quite perfect in this scenario because of the simplistic centering which does not account for the width differences of the 't' and the 'm' characters at the ends.

danmarshall, 2016-09-30:
I've just checked in a fix for the character width issue, reload and give it a try.

davejack1, 2016-10-03:
Thanks a lot!

My solution for auto-adjusting in accordance with the text length is the following;

`LibraryObject.curvedText = function (font, text, fontSize, arcRadius, onTop) {

            //generate the text using a font
            var textModel = new makerjs.models.Text(font, text, fontSize, false, true);

            //measure height of the text from the baseline
            var measure = makerjs.measure.modelExtents(textModel);
            var height = measure.high[1];
            var h2 = height / 2;
            var left = measure.low[0];
            var right = measure.high[0];
            var textWidth = right - left;

            //Calculate the angles dynamically
            var alpha = (textWidth*360)/(2*Math.PI*arcRadius);
            var startAngle = (180 - alpha)/2;
            var endAngle = (alpha + (180 - alpha)/2);

            var arc = new makerjs.paths.Arc([0, 0], arcRadius, startAngle, endAngle);

            //save all of these in the model
            this.models = {text: textModel};

            //move each character to a percentage of the total arc
            var span = makerjs.angle.ofArcSpan(arc);
            for (var i = 0; i < text.length; i++) {
                var char = textModel.models[i];

                //get the x distance of each character as a percentage
                var distFromFirst = char.origin[0] - left;
                var center = distFromFirst / textWidth;

                //set a new origin at the center of the text
                var o = makerjs.point.add(char.origin, [0, h2]);
                makerjs.model.originate(char, o);

                //project the character x position into an angle
                var angle = center * span;
                var angleFromEnd = onTop ? endAngle - angle : startAngle + angle;
                var p = makerjs.point.fromAngleOnCircle(angleFromEnd, arc);
                char.origin = p;

                //rotate the char to 90 from tangent
                makerjs.model.rotate(char, onTop ? angleFromEnd - 90 : angleFromEnd + 90, p);
            }

            return textModel
        };`

closed 2016-10-04 by davejack1

danmarshall, 2017-03-30:
FYI these are now built-in to makerjs.layout:
http://microsoft.github.io/maker.js/playground/?script=text-on-path
http://microsoft.github.io/maker.js/playground/?script=text-on-chain