microsoft/maker.js #212: Bounding box dimensions

davejack1, 2016-10-05:
Is it possible to find the bounding box dimensions of the geometry defined as combination of paths and models?

`var finalModel = {paths:{},models:{}};`

danmarshall, 2016-10-05:
Yes, see http://microsoft.github.io/maker.js/docs/basic-drawing/#Measuring

davejack1, 2016-11-04:
I tried to use that method but I am having trouble;

I keep all the paths I draw inside an array and at the end I put them into the `finalModel` variable, which will later contain some models, such as text.

    for (var i = 0; i<path.length; i++)
        {
            finalModel.paths[i] = path[i];
        }

When I try to get the bounding box of the whole path cluster, function returns null for high and low values;

var m = makerjs.measure.pathExtents(finalModel.paths);

But if I try it with individual paths `var m = makerjs.measure.pathExtents(finalModel.paths[0]);` it does work.

How can I measure the bounding box of the new shape made of group of paths?

danmarshall, 2016-11-04:
```makerjs.measure.pathExtents()``` will only accept a single path as a parameter. To measure a model, use ```makerjs.measure.modelExtents()``` there is no operator for working with arrays of paths.

danmarshall, 2016-11-04:
You can have a for...loop over your array and use [measure.increase](http://microsoft.github.io/maker.js/docs/api/modules/makerjs.measure.html#increase) to compound measurements. This is what modelExtents does anyways.

closed 2016-11-30 by danmarshall