microsoft/maker.js #213: Dimensions do not register

davejack1, 2016-10-14:
I am defining a variable like;

`var finalModel = {paths:{},models:{}};`

Setting the unit as milimeter

`finalModel.units = makerjs.unitType.Millimeter;`

And then I am generating some paths and pushing them into an array called `path`. And then I am assigning those paths to finalModel object's paths

    for (var i = 0; i<path.length; i++)
            {
                finalModel.paths[i] = path[i];
            }

Then returning that finalModel object and using it in a promise to add some text objects into it. And eventually I am converting it into DXF and saving it as a DXF file.

Then I am measuring a distance, which should be 6 Millimeters, but it is 6 inches instead.

Where am I doing wrong?

I also tried to define the finalModel variable as 

`var finalModel = {paths: path, units:makerjs.unitType.Millimeter}`

This didn't work as well.

danmarshall, 2016-10-14:
Inch is the default unit for DXF when there are no units specified. But it looks like you are specifying mm correctly. Can you share the JSON of your final model object ?

davejack1, 2016-10-14:
I don't know how to generate a JSON out of my model object?

danmarshall, 2016-10-14:
in your case it's ```finalModel``` - can you share the line of code where you execute ```makerjs.exporter.toDXF(...)```

davejack1, 2016-10-14:
var makerjs = require('makerjs');
var zip = new JSZip();

    var finalModel = {paths:{},models:{}};
    finalModel.units = makerjs.unitType.Millimeter;

    finalModel.models[0] = new makerjs.models.Square(100);

    function download() {
        zip.file("nested/" + '1' + ".dxf", makerjs.exporter.toDXF(converted));

        zip.generateAsync({type: "blob"})
            .then(function (blob) {
                saveAs(blob, "hello.zip");
            });

    }

    document.getElementById('save-btn').addEventListener('click',download);

One note; I divided every single variable by 25.4 to convert the whole drawing to millimeters. That worked well when I measure the distances on a DXF viewer (https://a360.autodesk.com/viewer/). However, I imported this DXF to a laser cutter software (I don't know the name of the software) and the model was very tiny there. Then I tried to save it with the default unit of Inches and then imported it into the laser cutter software and the dimensions were reasonable. It definitely was not inches, but didn't have the chance to measure any distance on the software.

danmarshall, 2016-10-14:
I see ```makerjs.exporter.toDXF(converted)``` - where is the ```converted``` coming from?

davejack1, 2016-10-15:
Oh sorry this line was from previous attempt. Is should be `makerjs.exporter.toDXF(finalModel)`

danmarshall, 2016-10-15:
Also, if your model does not have units, then you can assign units at the time of exporting:

```makerjs.exporter.toDXF(model, {units:makerjs.unitType.Millimeter})```

http://microsoft.github.io/maker.js/docs/exporting/#DXF

davejack1, 2016-10-15:
That does not make any difference. I still measure 100" as the length.

What do you think about my findings in the previous post?

danmarshall, 2016-10-19:
@davejack1 Yes regarding the DXF file format your findings are correct.

closed 2016-11-30 by danmarshall