microsoft/maker.js #220: Link to external js file

3kmch, 2016-12-11:
Is it already possible to add an external file to the playground? This way I don't have to set up my own Maker.js platform. See also at OpenJSCAD:
http://openjscad.org/#http://www.thingiverse.com/download:164128 (STL) or
http://openjscad.org/#http://openjscad.org/examples/slices/tor.jscad

danmarshall, 2016-12-11:
It's definitely possible, but it also presents a security issue because the js files would execute. If someone created a link to a malicious script, it could eventually blacklist the Maker.js site. 

Generally my solution to this was to accept PRs and host the js here on the site. This would mean you would be sharing your code publicly. 

If folks don't want their code to be public, we could have something like: Don't immediately execute external scripts, and show an "I agree" button that the user accepts the risk.

3kmch, 2016-12-12:
Thank you for your reply. :)

Sharing it on your site is a good solution. But I assume this would be a solution that depends on your spare time, so perhaps not an instant one. ;-) 

So having the opportunity to add the js file directly would be awesome and having an [I agree] button sounds reasonable.

danmarshall, 2016-12-12:
Another option to be aware of, is that it is pretty easy to host your own copy of the site, here on GitHub:
1. Click the "Fork" button in the top right corner of the page. This will create your own copy of this repo.
2. Visit http://YOURUSERNAME.github.io/maker.js/demos/

danmarshall, 2016-12-13:
Ok, you can now specify an external script. You will see a warning screen, and the code will not auto-execute, you must click the Run button.

Example:
http://microsoft.github.io/maker.js/playground/?script=https://gist.githubusercontent.com/danmarshall/8eb60f82ca765f477f0bc3a73966317a/raw/ebfeb4fa2db63726f6e9d30580f1df9f83b54ca1/heart.js

3kmch, 2016-12-13:
Cool, that was quick. :)

I tried to refer to
http://microsoft.github.io/maker.js/playground/?script=http://3km.ch/makerjs/sleeve.js
but got an error (message in browser):

> Load module failure : Could not load script "http://3km.ch/makerjs/sleeve.js". Possibly a network error, or the file does not exist.

Message in console:

> XMLHttpRequest cannot load http://3km.ch/makerjs/sleeve.js. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://microsoft.github.io' is therefore not allowed access.

Do you just allow files from github?

danmarshall, 2016-12-13:
Actually it's the 3km.ch server which is not allowing the resource. See https://developer.mozilla.org/en-US/docs/Web/HTTP/Access_control_CORS 

In my example I used a GitHub Gist which was convenient, you may want to try it: https://gist.github.com/

3kmch, 2016-12-14:
OK, I git it; did it the GitHub way:

http://microsoft.github.io/maker.js/playground/?script=https://raw.githubusercontent.com/3kmch/makerjs/master/sleeve.js

It just works. Thank you very much. :)

danmarshall, 2016-12-14:
Sure :)
Also have you tried running it on your local machine?

3kmch, 2016-12-15:
No, not yet.

closed 2017-01-03 by danmarshall