microsoft/maker.js #223: Polygon Model like OpenJSCAD

VVZen, 2017-02-01:
Hi guys!
I'm used to the [polygon()](https://en.wikibooks.org/wiki/OpenJSCAD_User_Guide#Polygon) function in **OpenJSCAD** which accepts n points and creates a closed polygon between them. Here in makerjs polygon can only be used to create [regular polygons](https://en.wikipedia.org/wiki/Regular_polygon).
It would be great to have it here, too!
For now, I'm just creating lines one by one but it would be a lot more readable to create a single polygon object with n lines instead of creating n lines individually.
```

This:

var lineAB = new makerjs.paths.Line(A, B);
var lineBC = new makerjs.paths.Line(B, C);
var lineCD = new makerjs.paths.Line(C, D);
var lineDA = new makerjs.paths.Line(D, A);
```

_would become_

```
var polyABCD = new makerjs.models.Polygon(A, B, C, D);
```

Thanks

PS: this library is really great for laser cutter owners! ❤️

VVZen, 2017-02-01:
After searching a bit through the docs, I saw that _makerjs_ has the object implementation that I was looking for under and it's actually called [ConnectTheDots](https://microsoft.github.io/maker.js/docs/api/classes/makerjs.models.connectthedots.html
) (fun indeed!).
So I was able to do:
``` new makerjs.models.ConnectTheDots(true, [A, B, C, D]); ```

A more detailed example can be found here : https://microsoft.github.io/maker.js/docs/basic-drawing/#content .

I think this can be closed.
This library is really awesome! 😎 💯

danmarshall, 2017-02-02:
Yes you found the correct implementation, thanks for the feedback @VVZen ! Please feel free to use issues to share any other ideas you have. 🥇

closed 2017-02-02 by danmarshall