microsoft/maker.js #224: tiny gcode visualizer lib for tiny webserver

mkeyno, 2017-02-12:
sorry to open ticket for my question but actually I'm looking for light gcode visualizer or probably cad2gcode convertor in my webpage , but this webpage hosted on tiny ESP826612F module which is connected to Arduino GRBL gcode interpreter by serial line, so I should have great care about its weight , My plan is just connect to wifi module and open the browser and do designated CAM job.,As I'm not pro in client side script like JavaScript I was hopping could have good advise here, do you suggest to make some plugin from such library so I shouldn't have to download or call all library ?
thanks

danmarshall, 2017-02-12:
Hi @mkeyno, sounds like a unique situation. If you have size constraints on your Arduino device, I might suggest only extracting the active gcode from that device's web server, then pull that gcode into a web app on another server which has more resources. Then you can have a full featured app.

mkeyno, 2017-02-12:
thanks @danmarshall , actually the main problem is number of file dependency is lot and make webserver unstable and should consider, there is no internet connection and module is in access point mode, in this situation we gonna merge files in to the one, but still zipped file size is high and in many situation it is not fully loaded in client browser so I was thinking maybe it is good idea to make plugin from library so put the weight on browser and only call class instance in the program,
I might add I'm not good in JS and maybe not describe problem well

closed 2017-04-07 by danmarshall

z3dev, 2017-04-07:
There's a GCODE visual piece in OctoPrint