microsoft/maker.js #229: Install web version with npm? Support for es6 import method?

sloba88, 2017-03-16:
Is it possible to provide support for importing plugin with npm using es6 style?

I am having hard times using this in es6 (webpack and babel) with import command with browser application. If I install with npm and try to import that it complains about the detectEnvironment.

What are the reasons behind using this require method?

danmarshall, 2017-03-16:
The browser version is created using Browserify, which is similar to Webpack. So you will want to webpack the node version. Can you tell me the error you see about detectEnvironment?

Or you can just use the already browserified version and treat it like an external library in your webpack config. You would just load with 2 scriot tags.

sloba88, 2017-03-16:
The current documentation for browser suggest pulling it from third party server and the file size is currently 333kb which is a lot.

So I believe installing browser version with some package manager would be great. Since bower is out of use and everyone use npm or yarn it would be good to put it in npm (or make it work with current npm setup you have). Then next step would be a possibility to import it in es6 way.

I could put it as external library and that is how I did it currently, and I also had to add bezier.js this way.

This is the error I get when I try to include the node version. It complains about this being undefined
[image]

danmarshall, 2017-03-16:
I'm not sure what you mean by 'put it in npm' - it already is:
```npm install makerjs```

I'm unable to repo your error, can you share a minimal example of your build?

danmarshall, 2017-03-16:
Oh, I see what you are asking, to add browser.maker.js to the npm. I can do that.

danmarshall, 2017-03-17:
Ok, you can now find the browser version in npm: node_modules\makerjs\target\js\browser.maker.js

danmarshall, 2017-03-17:
Also, I ran webpack and was not able to get this error, it returned "browser" correctly. What webpack config are you doing?

sloba88, 2017-04-07:
It still doesn't work for me. Now you have added the browser version to the npm but when I import with javascript first I have to use:

    import makerjs from './node_modules/makerjs/target/js/browser.maker.js'

which is not good, and then the importing this way is not the same as
```
<script src="./scripts/maker.js" type="text/javascript"></script>
<script type="text/javascript">
   var makerjs = require('makerjs');
</script>
```
The makerjs variable differs. So I can't really use the plugin with es6 import statement, right? 

What I want is to be able to do:
```
npm install makerjs
```
```
import makerjs from 'makerjs'
```

and that should be all. Is this unclear or unusual in any way? I don't think that webpack config has anything to do with this but if you still want I can show it.

danmarshall, 2017-04-07:
Can you create a small repo with the same configuration? This would make it easier to debug.

sloba88, 2017-04-10:
Yes, sure, here is the repo https://github.com/sloba88/makerjs-import

I believe this is the standard webpack setup and in my opinion import should be enough to import the library.

danmarshall, 2017-04-11:
Ok I've published changes in a new version 0.9.47, tested and it works with your repo's config.
BTW I think it already worked in Webpack 2, so I think the key info I needed to debug for you was that you were on Webpack 1. Let me know how it works for you.

sloba88, 2017-04-11:
I've just tried it and it works! Thanks! I will close the issue.

closed 2017-04-11 by sloba88