microsoft/maker.js #232: Unknown error with model.moveRelative()

microSoftware, 2017-03-28:
Hello, 

I've found a strange error when I want to move relatively a model. I run maker.js on nodejs 7.2.1

makerjs.model.moveRelative(underlineModel,  [20, **-20**]);
Make the process always running. So the request never end.

But with a positive y it work
makerjs.model.moveRelative(underlineModel,  [20, **10**]);

And with a negative x it work as well 
makerjs.model.moveRelative(underlineModel,  [-20, 20]);

So I really don't know what can be this error. And yeah, there is no error in the console nodejs. So even harder to debug.

If someone can try model.moveRelative with a negative y and tell me if it's just me or a bug in makerjs.

Thank you
PS: I've just discovered Maker.js and it's so well designed. +1 for the team

microSoftware, 2017-03-28:
Lol, when creating a Rect model. I used "50" instead of just 50 for the width.
Problem fixed !

closed 2017-03-28 by microSoftware

danmarshall, 2017-03-28:
Thanks a lot @microSoftware ! Please let me know any other feedback you have, I'm planning on a few breaking changes for 1.0.