microsoft/maker.js #233: [Question] How to import a svg or color/stroke a model ?

microSoftware, 2017-03-29:
Hello, 

I want to color and stroke several models. I haven't seen any other way to do it properly. So I end up doing it when I export the model. But the problem is I have several model with different colors. So I have to export each of the models : 
```
var textSvg = makerjs.exporter.toSVG(textModel, {
				"fill": textColor,
				"strokeWidth": 0
			});

var text2Svg = makerjs.exporter.toSVG(text2Model, {
				"fill": text2Color,
				"strokeWidth": 0
			});
```
So I have 2 svg generated. With the good color and the good stroke. Now I would like to merge these 2 svg.

I thought about using the method : fromSVGPathData() but I don't have the path but the whole svg. And even If I extract the path from the svg I will lose the style (color, stroke...) so not the good solution. 

Any idea ? 
Thank you for you help

danmarshall, 2017-03-29:
Do you need the colors inlined in the SVG? Can you use a CSS stylesheet in your scenario?

microSoftware, 2017-03-29:
Yep that's the things I need inline style. Just fill and stroke actually.

danmarshall, 2017-03-29:
Layers would be the solution for this. I would need to add some code to support this. but what you would end up with is something like this:

```
textModel.layer = 'layer1';
text2Model.layer = 'layer2';

var layerOptions = {
  "layer1": {
    "fill": textColor,
    "strokeWidth": 0
  },
  "layer2": {
    "fill": text2Color,
    "strokeWidth": 0
  }
};

var svg = makerjs.exporter.toSVG(textModel, {
				"layerOptions": layerOptions
			});
```

danmarshall, 2017-03-29:
Basically, we need this line:
https://github.com/Microsoft/maker.js/blob/master/src/core/svg.ts#L588

To look like this line:
https://github.com/Microsoft/maker.js/blob/master/src/core/svg.ts#L449

but it would get the `{ fill, stroke, etc.}` from a layerOptions that may be passed.

microSoftware, 2017-03-29:
Thanks for your reply. what the best way for me to modify the code ? 
To apply what you said I modified the js code from the node_module/makerjs/index.js
That the js code I modified is : 
```
//before
                     //var layerGroup = new exporter.XmlTag('g', { id: layer });
 
                     //after
                    var svgGroup = new exporter.XmlTag('g', {
                        id: 'svgGroup',
                        stroke: layer.stroke,
                        "stroke-width": layer.strokeWidth,
                        "stroke-linecap": "round",
                        "fill": layer.fill,
                        "fill-rule": "evenodd",
                        "font-size": layer.fontSize
                    });
```
It doesn't work but we move forward.
To get the stroke/fill ... from the layer.
I modified all the opts.fill => layer.fill  (doesn't work)
And I tried layer.opts.fill (doesn't work)

Do you know how I get the options of the layer ?

danmarshall, 2017-03-29:
Actually I think it's even easier. Revert your changes in the js code :)
Then add a layer property to each model, then export as before.
Each layer will become a new '<path>' element in your SVG with the id of the layer. Then just manipulate the XML dom conventionally, or use string.replace to inject your additional attributes.

microSoftware, 2017-03-30:
I'm not sure if I completely get what you said. But I set a layer for each model. The id is not added on the path nor on the group. 
So what I end up doing is : 
```
textModel.layer = "text";
			text2Model.layer = "text2";
			
			var options = {
				"text":{
					"fill":'red',
					"stroke-width":'0'
				},
				"text2":{
					"fill":'blue',
					"stroke-width":'1',
					"stroke":"green"
				}
			}

			var models = {
				"models":{
					"text":textModel,
					"text2":text2Model
				}
			}

			var svg = makerjs.exporter.toSVG(models, options);
```

And I added a small function in makerjs to create a group
```
function createGroup(value, options){
                var optionString = "";
                for(option in options){
                    var attr = option;
                    var valueOption = options[attr];
                    optionString += attr + '="' + valueOption + '" ';
                }
                 var group = "<g "+optionString+">" + value + "</g>";
                 return group;
            }
```

And modified the append function : 
```
 if(layer != undefined){
                         var group = createGroup(value, options[layer]);
                         elements.push(group);
                    }
                    else{
                         elements.push(value);
                    }
```

Not the most beautiful thing but that'll do it for the moment.
Thanks for you help 👍

danmarshall, 2017-03-30:
I just did this test, and got this result which doesn't completely give us everything:
```
var makerjs = require('makerjs');

//Square parameters: side
var square1 = new makerjs.models.Square(1);
var square2 = makerjs.model.move(new makerjs.models.Square(1), [2, 0]);

var model = {
  models: {
    square1: square1, square2: square2
  }
};

console.log(makerjs.exporter.toSVG(model));

/*
 <svg width="3" height="1" viewBox="0 0 3 1">
  <g id="svgGroup" stroke="#000" stroke-width="0.25mm" stroke-linecap="round" fill="none" fill-rule="evenodd" font-size="9pt">
   <path d="M 0 1 L 1 1 L 1 0 L 0 0 L 0 1 Z M 2 1 L 3 1 L 3 0 L 2 0 L 2 1 Z" vector-effect="non-scaling-stroke"/>
  </g>
 </svg>
*/

//now add layers so each sub-model becomes a different <path>
square1.layer = "layer1";
square2.layer = "layer2";

console.log(makerjs.exporter.toSVG(model));

/*
 <svg width="3" height="1" viewBox="0 0 3 1">
  <g id="svgGroup" stroke="#000" stroke-width="0.25mm" stroke-linecap="round" fill="none" fill-rule="evenodd" font-size="9pt">
   <path d="M 0 1 L 1 1 L 1 0 L 0 0 L 0 1 Z" vector-effect="non-scaling-stroke"/>
   <path d="M 2 1 L 3 1 L 3 0 L 2 0 L 2 1 Z" vector-effect="non-scaling-stroke"/>
  </g>
 </svg>
*/
```

So 2 paths get outputted, but there is a feature missing, putting the layerID on the path.

microSoftware, 2017-03-30:
Yep you're right. 
I actually think a good idea would be : 
If the model has no layer: do the same way than before
If the model has a layer : 
1) Wrap the path in a <g> element
2) Put the id on the <g> element => id=layerName
3) Take an option parameter and apply the style on the g element.

On the issues board i've seen people who wanted to style their model but the only way was in css. This would be a really easy way to do it. Don't you think ?

If I fork the project to apply this modif. Should I fork 2 times to apply this modif and the other issue OR I can fork only once to fix both issues ?

danmarshall, 2017-03-30:
One fork should do.

Yes pretty close to what I was thinking too.

And, I was incorrect about the line of code, it's: 
https://github.com/Microsoft/maker.js/blob/master/src/core/svg.ts#L466

I think the `<g>` may not be necessary, we can probably just put the id on the `<path>` but I'm open.

danmarshall, 2017-03-30:
also, the `vector-effect` would probably be an option too.

danmarshall, 2017-04-07:
Hi @microSoftware , I added id to paths by layer, and it's been released in 0.9.46

closed 2017-04-07 by danmarshall