microsoft/maker.js #234: Suggestion : add opentype option argument

microSoftware, 2017-03-30:
Hello, 

I'd look to modify the letter-spacing, kerning...when creating a Text model. I tweak the code but thought it would be cool to integrate it officially.

Text.ts : https://github.com/Microsoft/maker.js/blob/master/src/models/Text.ts#7
`constructor(font: opentype.Font, text: string, fontSize: number, combine = false, centerCharacterOrigin = false, bezierAccuracy?: number, **optionsOpenType**) `

And modify https://github.com/Microsoft/maker.js/blob/master/src/models/Text.ts#95
by `font.forEachGlyph(text, 0, 0, fontSize, **optionsOpenType**, cb);`

Then when creating a model we can add the opentype option : 
```
{
                letterSpacing: -0.1,
                kerning: true 
}

```

I've seen https://github.com/Microsoft/maker.js/blob/master/src/models/Text.ts#L12
there is already a parameter options: opentype.RenderOptions
But I don't understand why it is there.

If someone can give me his thought on this.

danmarshall, 2017-03-30:
Great idea! Glad to see you using the library to the fullest. Feel free to submit a PR to the dev branch.

The options param on the `cb` function is just to satisfy TypeScript saying that this callback function implements the signature with all specified parameters that Opentype expects.

microSoftware, 2017-03-30:
Ok I do that. Like that It will give the opportunity to try typescript on the server side.

danmarshall, 2017-04-07:
Hi @microSoftware I went and added this and released 0.9.46, since I saw you had several other changes in your fork.

closed 2017-04-07 by danmarshall