microsoft/maker.js #239: 1x1 Ellipses constructed with multiple arguments causes errors

petestreet, 2017-04-03:
When creating an Ellipse, it's often the case that `MakerJs.path.distort()` is called. If the ellipse's dimensions are 1x1 (i.e. a circle), the `return null` [here](https://github.com/Microsoft/maker.js/blob/master/index.js#L1061) will cause an error to bubble up when we pass `null` into the `BezierCurve()` constructor ([here](https://github.com/Microsoft/maker.js/blob/master/index.js#L7142)). 

For our use-case, since we aren't also using `EllipticArc` models, it was enough to comment out that `return null`. But there should be a general way to not distort in the case of 1x1 ellipses.

danmarshall, 2017-04-04:
Wow good catch, thanks for pointing this out. This was supposed to be an optimization, but now I see that it should actually return a clone in the 1x1 case.

danmarshall, 2017-04-05:
Fixed in 0.9.45

closed 2017-04-05 by danmarshall