microsoft/maker.js #240: Weird bug with the evenodd property on Text

microSoftware, 2017-04-04:
Hello, 

The default property of fill-rule is evenodd. Which should be ok.
But when I convert a text in path using the model.Text I got something really strange : 
[image]

I don't want the intersection of the letter to be white. I want them to be the same color.

I tried with many different font. Same problem.
Using the fill-rule="nonzero" make the problem even worst.

Any idea how I can fix that ?
Thanks

danmarshall, 2017-04-04:
Set the "combine" parameter to true in makerjs,models.Text(font, text, fontSize, combine, centerCharacterOrigin, bezierAccuracy)
http://microsoft.github.io/maker.js/docs/api/classes/makerjs.models.text.html#constructor

Here is an example, try checking / unchecking the "combine" checkbox:
http://microsoft.github.io/maker.js/playground/?script=Text#params=["montserratsubrayada-bold","Hello",72,true,false]

microSoftware, 2017-04-05:
Thanks. Indeed. combine property solved the problem.

I have another weird behavior though. For most of the cursive/script fonts : 
With cursive/script font i have some weird behaviors. See image below : 
text : cde => GOOD
[image]

text : acde => NOT GOOD. 
[image]

text : aac => NOT GOOD
[image]

First I thought It was a opentype.js problem. But I display the text using only opentype.js and it work. So it's probably in makerjs.
I think the problem came from the way the glyph are transform in model  with font.forEachGlyph(text, 0, 0, fontSize, opentypeOptions, cb); 

Any idea ? else I would just rewrite another function using getPath method from Opentype. And then import the path in the model. It's not the idea of Makerjs but It should fixe the bug and convert it faster 
```
var pathObj = font.getPath(word, 0, 0, 40);
var pathData = pathObj.toPathData(1);
```

Thanks for your help

danmarshall, 2017-04-05:
This may be due to a bug in bezier.js not calculating arcs correctly, can you provide the JSON for just the "a" character?

microSoftware, 2017-04-06:
Ok. Not all the times, but sometimes in the console I have this message "arc abstraction somehow failed..."

I attached 2 json : 
- one that convert only the character 'a' and this one is good (a_good.json)
- one that convert "aac" and the letters are not displayed correctly. (a_not_good.json)

[Archive.zip](https://github.com/Microsoft/maker.js/files/901145/Archive.zip)

danmarshall, 2017-04-06:
The bezier.js library has been updated just now, can you try again? You may need to git pull.

microSoftware, 2017-04-06:
I pull the whole project maker.js. But same problem. I've seen makerjs use bezier-js 2.1.0. 
Is there a compatibility reason to not move to bezier-js 2.2.x ?

danmarshall, 2017-04-06:
Oh, yes you are right, the node version should also be published. I updated the browser version :|

microSoftware, 2017-04-06:
I updated bezier-js to 2.2.3. Still the problem. I've seen there is bug for more simple font as well (mosk.ttf) . For example, the middle of the "o" will be fill out.

microSoftware, 2017-04-06:
If I generate the text outside makerjs and import it in makerjs the problem will also occur. Is there a way to import a brute raw path without convert into a makerjs model ? (letting the positioning problem aside)

danmarshall, 2017-04-06:
Yes I am also seeing some other bezier errors. Seems worse than before. Perhaps you can downgrade for the moment.

danmarshall, 2017-04-06:
There was a pull request that may have fixed the problem: https://github.com/deanmartinez/bezierjs/blob/4cbbc9223d5159d8796557163032aaea14bf602b/lib/bezier.js

Maybe you can try to put this into your node_modules ?

microSoftware, 2017-04-06:
Nice it fixed both bugs 👍

danmarshall, 2017-04-06:
Awesome! I will wait for the lib to stabilize before updating.

microSoftware, 2017-04-06:
okay. Should I close the issue or leave it open until the update ?

danmarshall, 2017-04-06:
Let's leave it open in case someone else has an issue, they will see it here. Thanks!

microSoftware, 2017-04-06:
Actually I have still the problem. Just less frequently. Which is still good though.

danmarshall, 2017-04-06:
ok try and get specific repro steps if possible.

microSoftware, 2017-04-06:
Ok I do that asap. When i import a path with the method makerjs.importer.fromSVGPathData() the intersection between each path are not filled up. Is there a way to modify this behaviour the same way than with the argument combine of Text ?

danmarshall, 2017-04-06:
Doing it that way is going to be tricky, because you will not know which paths make up a letter. You will be wanting to combineUnion each letter, but you will also be not wanting to combineUnion the "hole" in the middle of a letter like o. You can try to findChains but since your letters are already overlapping, there is a small chance that you may get a "figure 8" kind of chain if they have paths that share a point.

danmarshall, 2017-04-06:
If you are willing to have a very small outline surrounding your shape, you may want to look at doing something like `model.expandPaths(yourModel, .002)` and then doing a `model.findLoops` and removing loops by their order of containment.

microSoftware, 2017-04-06:
OK. I made a small repo to test easily opentype and makerjs text : 
1) git clone https://github.com/microSoftware/display-font.git
2) npm install 
3) node makerjs.js (or nodemon makerjs.js if you want to watch for modif)
4) modify the words to see which one work and don't work
5) Go on http://localhost:8900 to see the result

Note 1 : makerjs used is the fork I made. But I almost didn't change anything. And it's the last version of makerjs

Note 2 : when you will run npm install, you need to modify the bezierjs/lib/bezierjs with  https://github.com/deanmartinez/bezierjs/blob/4cbbc9223d5159d8796557163032aaea14bf602b/lib/bezier.js because the modifs will have disapeared.

Tell me if you have any issues testing. You can test before and after modifying bezierjs. It's not the same words that don't work. And changing the font as well.

microSoftware, 2017-04-07:
@danmarshall Thanks for your answer. I think it's going to be a bit slow to use findLoop(). My dream solution would be to use opentype.js font.getPath. Because it's freaking fast. But I just can't figure out how to get the value of x and y based on the font and text.

danmarshall, 2017-04-07:
Yes, boolean combination is not a fast operation ;)

microSoftware, 2017-04-07:
I thought it was going to loop through every point of the whole path. Which would be long with a cursive font. But I'll give it a try.It's the best way to see what the result and performance looks like.

danmarshall, 2017-04-14:
Hi @microSoftware thanks for your repo - I was able to reproduce your scenario. I have some more testing to do, but here is a preliminary patch of `isBetweenPoints` you may like to try:
```
        function isBetweenPoints(pointInQuestion, line, exclusive) {
            var oneDimension = false;
            for (var i = 2; i--;) {
                if (MakerJs.round(line.origin[i] - line.end[i], .000001) == 0) {
                    if (oneDimension)
                        return false;
                    oneDimension = true;
                    continue;
                }
                var origin_value = MakerJs.round(line.origin[i]);
                var end_value = MakerJs.round(line.end[i]);
                if (!isBetween(MakerJs.round(pointInQuestion[i]), origin_value, end_value, exclusive))
                    return false;
            }
            return true;
        }
        measure.isBetweenPoints = isBetweenPoints;

```

microSoftware, 2017-04-18:
I test with node.js and got an error : 
```
/Applications/MAMP/htdocs/display-font/node_modules/bezier-js/lib/utils.js:540
      arc.s = s;
            ^

TypeError: Cannot create property 's' on boolean 'false'
    at Object.getccenter (/Applications/MAMP/htdocs/display-font/node_modules/bezier-js/lib/utils.js:540:13)
    at Bezier._iterate (/Applications/MAMP/htdocs/display-font/node_modules/bezier-js/lib/bezier.js:797:23)
    at Bezier.arcs (/Applications/MAMP/htdocs/display-font/node_modules/bezier-js/lib/bezier.js:758:19)
    at getArcs (/Applications/MAMP/htdocs/display-font/node_modules/makerjs/index.js:6845:26)
    at new BezierCurve (/Applications/MAMP/htdocs/display-font/node_modules/makerjs/index.js:7016:32)
    at /Applications/MAMP/htdocs/display-font/node_modules/makerjs/index.js:7003:64
    at Array.forEach (native)
    at new BezierCurve (/Applications/MAMP/htdocs/display-font/node_modules/makerjs/index.js:7002:36)
    at /Applications/MAMP/htdocs/display-font/node_modules/makerjs/index.js:8077:42
    at Array.map (native)
```
I modified bezier.js with the code you gave me last time. Should I put the original code back ?

danmarshall, 2017-04-18:
That happens when a Bezier is actually a straight line. Can you provide a repro?

microSoftware, 2017-04-20:
1.  git clone https://github.com/microSoftware/display-font.git
2. cd display-font 
3. npm install
4. node .

The error occurred only with some words. For example : "okitlaks" with the font "galaxie_cassiopeia_bold"

danmarshall, 2017-04-20:
Thanks, was able to repro. Will work on a fix.

danmarshall, 2017-04-20:
Ok I believe this is fixed - can you update to 0.9.49 and verify?

microSoftware, 2017-04-21:
Nice job !!! Thank you 😀  😀  😀

closed 2017-04-21 by danmarshall