microsoft/maker.js #246: Ability to add tabs for boxes

objectiveSee, 2017-04-17:
A common use for laser cutting is to make boxes. A box can be made by create "tab" or fingers that allow different sides of the box to slot together. It would be great to add some code to this repo that would let us add slots to a path or shape. For example, like what http://www.makercase.com/ allows you to do.

[image]

danmarshall, 2017-04-17:
Absolutely! I am intending on doing this, just need to find the time. Meanwhile there is an example in http://microsoft.github.io/maker.js/docs/api/modules/makerjs.layout.html#clonetorow - just click the "Play" button. I just realized how buried this example is :(

danmarshall, 2017-04-17:
Ok, here you go: http://microsoft.github.io/maker.js/playground/?script=tongue-and-groove

objectiveSee, 2017-04-18:
Interesting example @danmarshall. I ended up implementing this myself in a module that accepts a model and traverses it path by path to add a notch on each line segment. Still figuring out the use cases that I have for this project.

danmarshall, 2017-04-18:
Is it private or can you share your example? Would like to see it :)

objectiveSee, 2017-04-18:
I can share it once I am in a working state again. Just ran into a bug. I am trying to expand a model using outline(), then originate() and simplify() and then add notches. What I am seeing however is that the direction of some lines is getting reverse at some step along the way. For example, a line that used to go from point A to point B now goes from point B to point A. It renders the same visually, but my logic was relying on direction of lines not changing as I am using connectTheDots() to product the final notched model. Hope that makes sense.

danmarshall, 2017-04-18:
Oh interesting. Are you using findChains() ?

objectiveSee, 2017-04-18:
No, I am not aware of that API. I was using the following code to traverse the paths. 

```
var keys = Object.keys(model.paths)
keys.forEach(function(path_name) {
   var path = model.paths[path_name]
   // and so on
})
```

danmarshall, 2017-04-18:
Take a look at http://microsoft.github.io/maker.js/docs/advanced-drawing/#Chains
There is also [findsinglechain](http://microsoft.github.io/maker.js/docs/api/modules/makerjs.model.html#findsinglechain) which is easier to call if you only have one chain and one layer.

danmarshall, 2017-04-18:
And for traversal there is http://microsoft.github.io/maker.js/docs/api/modules/makerjs.model.html#walk 
probably another thing which could use better documentation...

objectiveSee, 2017-04-18:
@danmarshall I think i can get this to work, but I did notice that sometimes the origin/end points of a path are reversed. You can use this GIST as an example: https://gist.github.com/objectiveSee/02c24932ef2258b99e1a0361517dd704

I added a log to check whenever the origin of a path does not equal the `end` of the previous path. You can see that this happens sometimes in the console where the end and origin are flipped. Not sure if thats a bug or not, but it goofed me up ;)

Some output:
```
----
ShapeLine1_Top Origin: (-0.32,9.32) End:(0.68,9.32)
----
ShapeLine2_Top Origin: (0.68,9.32) End:(0.68,10.32)
----
ShapeLine2_EndCap_ShapeLine1 Origin: (0.68,10.32) End:(1.68,10.32)
----
ShapeLine4_Top Origin: (1.68,10.32) End:(1.68,11.32)
----
ShapeLine4_EndCap_ShapeLine1 Origin: (1.68,11.32) End:(5.32,11.32)
----
ShapeLine5_EndCap_ShapeLine1 Origin: (5.32,10.32) End:(5.32,11.32)
:: warning :: path is broken.(5.32,11.32) should equal (5.32,10.32)
----
ShapeLine7_Top Origin: (5.32,10.32) End:(7.68,10.32)
----
ShapeLine8_Top Origin: (7.68,10.32) End:(7.68,11.32)
----
ShapeLine8_EndCap_ShapeLine1 Origin: (7.68,11.32) End:(11.32,11.32)
----
ShapeLine9_EndCap_ShapeLine1 Origin: (11.32,10.32) End:(11.32,11.32)
:: warning :: path is broken.(11.32,11.32) should equal (11.32,10.32)
----
```

objectiveSee, 2017-04-18:
Ok, notching is working now! Here is the working code to paste into the playground. There is a hard-coded "pixel heart" model which I then expand using outline() and then add notches to. Both models shown in the playground output.

https://gist.github.com/objectiveSee/24cabe403c60266ecfed2feae262358b

danmarshall, 2017-04-18:
Nice! That is impressive =)
Does it only work with "squared" shapes?

danmarshall, 2017-04-18:
So, were you expecting that this code would traverse your paths in order?
```var keys = Object.keys(model.paths)
keys.forEach(function(path_name) {
   //these will not be in order
})
```
Because, it sometimes works, but order isn't guaranteed: http://stackoverflow.com/questions/5525795/does-javascript-guarantee-object-property-order

This got me too. The only time order is guaranteed is in an Array. But I don't recommend using arrays as MakerJs objects.

objectiveSee, 2017-04-18:
I was expecting that order to be true (assuming that the keys were in order based on how they were named or generated). Does `makerjs.model.walk` guarantee that the order of traversing the paths is continuous? 

The actual bug I ran into was about `end` and `origin` being opposite of what I had expected. This seems to only happy when the x or y movement is in the negative direction. For example in `ShapeLine5_EndCap_ShapeLine1 ` above, its start point should have been `(5.32,11.32)` since that was the `end` of the previous line, `ShapeLine4_EndCap_ShapeLine1 `.

objectiveSee, 2017-04-18:
@danmarshall Yes, my notching algorithm only works on straight lines. I took some shortcuts since my design allowed for it.

danmarshall, 2017-04-19:
No, `makerjs.model.walk` does not guarantee order, but .makerjs.model.findChains' does (also `makerjs.model.findSingleChain`), in its `links` property which is an array.

danmarshall, 2017-04-19:
Whoa let me back up - the order is not the order of your objects keys. The order is in how the paths connect to each other. If your keys are sortable, then you might just sort them as an array then loop over that array. Watch out for things like `p11` coming prior to `p2` when sorting strings.

objectiveSee, 2017-04-19:
Thanks for the heads up on `p11` bug. Chains are the way to go, and these two look like they will get me all the way there:
```
toKeyPoints(chainContext: IChain, maxArcFacet?: number): IPoint[]

toPoints(chainContext: IChain, distanceOrDistances: number | number[], maxPoints?: number): IPoint[]
```

danmarshall, 2017-05-14:
Hi Danny, I'm going to close this for now, please re-open if there is anything else you'd like to add.

closed 2017-05-14 by danmarshall

objectiveSee, 2017-05-15:
Thanks @danmarshall! I did find the Joinery tool (https://clementzheng.github.io/joinery/) which might be promising for others to try.