microsoft/maker.js #248: Ability to set colors per model or layer

objectiveSee, 2017-04-18:
A common situation for laser cut designs is to have the "cut" lines be red and the "etch" lines to be blue. It would be nice if we could specify multiple colors when exporting to SVG so that we don't have to set the colors manually from another app after rendering the SVG. Setting the color on a model or layer should be adequate.

danmarshall, 2017-04-18:
This will fit in nicely with an upcoming change I need to make to be able to set other export options by layer.

danmarshall, 2017-04-19:
FYI there is also color using CSS - not sure if your SVG scenario is web based or not:
https://github.com/Microsoft/maker.js/issues/185

objectiveSee, 2017-04-21:
Thanks Dan, my scenario is all node based currently, plus I'm terrible with CSS, so I'll just wait on this feature. Luckily I only need to change the color of two layers so it's a simple step in Illustrator.

PR #254 (2017-05-14): new build 0.9.51
Added support for layer options in SVG export.
Fixes #248
closed 2017-05-14 by danmarshall