microsoft/maker.js #250: Help with combine() function

objectiveSee, 2017-04-24:
Hello. I am trying to use the combine() function to add a rectangle to a complex model. I have not bee able to configure the options of the function correctly to get the effect I am looking for. Specifically, I would like to extend the "pixel heart" model of mine by adding a rectangle. I tried using combine() and combineUnion() but haven't been able to get it to work. I did notice that the location where the rectangle is supposed to intersect my model has been broken into 3 paths by combine() (see screenshot). Any ideas as to what I am doing wrong? Thanks :1st_place_medal: 

Here is my code that runs in the playground:
```
var makerjs 			= require('makerjs')
var model = {"paths":{"ShapeLine1":{"type":"line","origin":[0,9],"end":[1,9]},"ShapeLine2":{"type":"line","origin":[1,9],"end":[1,10]},"ShapeLine3":{"type":"line","origin":[1,10],"end":[2,10]},"ShapeLine4":{"type":"line","origin":[2,10],"end":[2,11]},"ShapeLine5":{"type":"line","origin":[2,11],"end":[5,11]},"ShapeLine6":{"type":"line","origin":[5,11],"end":[5,10]},"ShapeLine7":{"type":"line","origin":[5,10],"end":[8,10]},"ShapeLine8":{"type":"line","origin":[8,10],"end":[8,11]},"ShapeLine9":{"type":"line","origin":[8,11],"end":[11,11]},"ShapeLine10":{"type":"line","origin":[11,11],"end":[11,10]},"ShapeLine11":{"type":"line","origin":[11,10],"end":[12,10]},"ShapeLine12":{"type":"line","origin":[12,10],"end":[12,9]},"ShapeLine13":{"type":"line","origin":[12,9],"end":[13,9]},"ShapeLine14":{"type":"line","origin":[13,9],"end":[13,6]},"ShapeLine15":{"type":"line","origin":[13,6],"end":[12,6]},"ShapeLine16":{"type":"line","origin":[12,6],"end":[12,5]},"ShapeLine17":{"type":"line","origin":[12,5],"end":[11,5]},"ShapeLine18":{"type":"line","origin":[11,5],"end":[11,4]},"ShapeLine19":{"type":"line","origin":[11,4],"end":[10,4]},"ShapeLine20":{"type":"line","origin":[10,4],"end":[10,3]},"ShapeLine21":{"type":"line","origin":[10,3],"end":[9,3]},"ShapeLine22":{"type":"line","origin":[9,3],"end":[9,2]},"ShapeLine23":{"type":"line","origin":[9,2],"end":[8,2]},"ShapeLine24":{"type":"line","origin":[8,2],"end":[8,1]},"ShapeLine25":{"type":"line","origin":[8,1],"end":[7,1]},"ShapeLine26":{"type":"line","origin":[7,1],"end":[7,0]},"ShapeLine27":{"type":"line","origin":[7,0],"end":[6,0]},"ShapeLine28":{"type":"line","origin":[6,0],"end":[6,1]},"ShapeLine29":{"type":"line","origin":[6,1],"end":[5,1]},"ShapeLine30":{"type":"line","origin":[5,1],"end":[5,2]},"ShapeLine31":{"type":"line","origin":[5,2],"end":[4,2]},"ShapeLine32":{"type":"line","origin":[4,2],"end":[4,3]},"ShapeLine33":{"type":"line","origin":[4,3],"end":[3,3]},"ShapeLine34":{"type":"line","origin":[3,3],"end":[3,4]},"ShapeLine35":{"type":"line","origin":[3,4],"end":[2,4]},"ShapeLine36":{"type":"line","origin":[2,4],"end":[2,5]},"ShapeLine37":{"type":"line","origin":[2,5],"end":[1,5]},"ShapeLine38":{"type":"line","origin":[1,5],"end":[1,6]},"ShapeLine39":{"type":"line","origin":[1,6],"end":[0,6]},"ShapeLine40":{"type":"line","origin":[0,6],"end":[0,9]}}}

var rectangle = new makerjs.models.Rectangle(1, 1)

makerjs.model.move(rectangle, [-.5,7]) // position so it overlaps off the left side by .5 
makerjs.model.combine(model, rectangle,true,true,false,true)

module.exports = model
```

and the screenshot:

[image]

danmarshall, 2017-04-24:
Thanks for the code snippet, that really helps. You just need to add the rectangle to your output model:
```
model.models = {
  myRectangle: rectangle
};
```

objectiveSee, 2017-04-24:
That worked! thanks for the quick reply. I came across combine() in the docs and realized it was going to do the box notching mentioned in story #246 much better than the solution I had. It is interesting that `rectangle` is also mutated by the combine operation.

closed 2017-04-24 by objectiveSee

danmarshall, 2017-04-24:
Yes the mutations are an interesting side effect. Let me know your feedback on that. It's primarily to keep the modeling in the same structure that you create.

danmarshall, 2017-04-24:
Check out the first few sections in http://microsoft.github.io/maker.js/docs/advanced-drawing/#content

objectiveSee, 2017-04-24:
After using `combine()` you might end up with a mix of paths and models within your main model. I could see that being annoying for some people. In that case however, would calling `simplify()` clear out the sub-models and simplify the model into just a set of paths?

danmarshall, 2017-04-24:
I tried to mimick Adobe Illustrator in the combine() function. Not sure if you use AI, but when you select 2 shapes, you can choose "combine" from their menu. This mutates both shapes into one. So that's how the mutation came to be.

It would be fairly simple to write a non-mutating wrapper. Here's the pseudo-code:

1. Clone modelA and modelB.
2. Perform combine on the clones.
3. Return a new model, which has 2 child models, cloneA and cloneB.

BTW combine does not create sub-models, but it does create more paths. Every path is broken into two, at the point where it intersects a path in the other model. These broken bits exist in the same place in the tree; so they need to have a unique but similar name.
Now everything is completely broken to the point where there are no intersections at all. Then the unwanted bits are removed, whether you wanted the inside or outside ones.

The simplify() function does the opposite of the above, it takes broken bits and reunites them. Well, it will unite any line on the same slope, or any arc on the same origin/radius. Perhaps it should be renamed...

But it does not do tree pruning, which may be what you are thinking of. After a combine, you may have removed entire swaths of a drawing leaving no renderable paths, but the tree structures remain. I was thinking of implementing a prune function to clean up trees after a combine, but this introduces other problems. There may be code which expected a certain structure to a given model. Typically it is ok to have a null at the end of a tree route, like model.models.foo.paths.bar - where bar is undefined, but if the route had a null along the way the code would fail and terminate.

danmarshall, 2017-04-25:
After reading your question again, you might be referring to a concept I call "flattening" - so there is not much of a tree, just one model with all paths. This is possible but there's also downsides. Having models within models is desirable because each sub-model can be measured as a group, and when doing further combine() operations it is helpful to cache the bounding box of each sub-model. Then, some intersections can be avoided altogether if we know they are outside of a bounding box.

objectiveSee, 2017-04-26:
Thanks for the clarification Dan. I think flattening for me would be nice as it simplifies the data schema, though I see your reason for not doing it by default. For my project I am all set currently. I was able to get everything working and am looking to move to the laser cutter this week for some prototypes.