microsoft/maker.js #268: Setting origin of imported paths causes script to hang

adamdyson, 2017-06-26:
I am trying to unite some SVG paths in a simple node application, which failed with an error relating to "Trimming Dead Ends". I can't post the specific error message and where it failed, because I've since simplified my code to try and debug the issue.

Please see code below which fails to simply import a couple of SVG paths and position them. Any help in figuring out the cause would be greatly appreciated.

```
	var data = {
		"options": {
			"bezierAccuracy":0.5
		},
		"paths":[
			{
				"x":"0", 
				"y":"72", 
				"d":"M 39.796875 -44.296875 C 39.796875 -43.285156 39.660156 -42.363281 39.390625 -41.53125 C 39.117188 -40.707031 38.75 -39.992188 38.28125 -39.390625 C 37.8125 -38.796875 37.238281 -38.332031 36.5625 -38 C 35.882812 -37.675781 35.144531 -37.515625 34.34375 -37.515625 C 33.4375 -37.515625 32.613281 -37.707031 31.875 -38.09375 C 31.132812 -38.476562 30.492188 -39.035156 29.953125 -39.765625 C 30.847656 -40.421875 31.5625 -41.289062 32.09375 -42.375 C 32.632812 -43.46875 32.90625 -44.632812 32.90625 -45.875 C 32.90625 -47.425781 32.445312 -48.648438 31.53125 -49.546875 C 30.625 -50.453125 29.375 -50.90625 27.78125 -50.90625 C 26.8125 -50.90625 25.921875 -50.722656 25.109375 -50.359375 C 24.304688 -49.992188 23.617188 -49.488281 23.046875 -48.84375 C 22.472656 -48.207031 22.023438 -47.441406 21.703125 -46.546875 C 21.390625 -45.660156 21.234375 -44.6875 21.234375 -43.625 C 21.234375 -42.03125 21.617188 -40.53125 22.390625 -39.125 C 23.171875 -37.71875 24.144531 -36.332031 25.3125 -34.96875 C 26.488281 -33.613281 27.753906 -32.238281 29.109375 -30.84375 C 30.472656 -29.457031 31.738281 -27.972656 32.90625 -26.390625 C 34.082031 -24.804688 35.054688 -23.09375 35.828125 -21.25 C 36.597656 -19.414062 36.984375 -17.382812 36.984375 -15.15625 C 36.984375 -12.789062 36.488281 -10.601562 35.5 -8.59375 C 34.519531 -6.59375 33.15625 -4.859375 31.40625 -3.390625 C 29.664062 -1.929688 27.617188 -0.785156 25.265625 0.046875 C 22.910156 0.878906 20.359375 1.296875 17.609375 1.296875 C 15.316406 1.296875 13.238281 0.992188 11.375 0.390625 C 9.507812 -0.203125 7.914062 -1.050781 6.59375 -2.15625 C 5.269531 -3.269531 4.242188 -4.597656 3.515625 -6.140625 C 2.785156 -7.691406 2.421875 -9.421875 2.421875 -11.328125 C 2.421875 -12.921875 2.675781 -14.40625 3.1875 -15.78125 C 3.707031 -17.164062 4.410156 -18.367188 5.296875 -19.390625 C 6.191406 -20.410156 7.25 -21.210938 8.46875 -21.796875 C 9.6875 -22.378906 10.988281 -22.671875 12.375 -22.671875 C 14.0625 -22.671875 15.476562 -22.25 16.625 -21.40625 C 17.78125 -20.5625 18.5 -19.40625 18.78125 -17.9375 C 17.507812 -17.9375 16.367188 -17.757812 15.359375 -17.40625 C 14.359375 -17.050781 13.503906 -16.554688 12.796875 -15.921875 C 12.097656 -15.296875 11.550781 -14.53125 11.15625 -13.625 C 10.769531 -12.71875 10.578125 -11.703125 10.578125 -10.578125 C 10.578125 -8.472656 11.191406 -6.804688 12.421875 -5.578125 C 13.660156 -4.347656 15.320312 -3.734375 17.40625 -3.734375 C 18.664062 -3.734375 19.832031 -3.957031 20.90625 -4.40625 C 21.988281 -4.863281 22.914062 -5.488281 23.6875 -6.28125 C 24.46875 -7.082031 25.078125 -8.035156 25.515625 -9.140625 C 25.960938 -10.242188 26.1875 -11.441406 26.1875 -12.734375 C 26.1875 -14.046875 25.96875 -15.300781 25.53125 -16.5 C 25.101562 -17.707031 24.523438 -18.878906 23.796875 -20.015625 C 23.078125 -21.160156 22.257812 -22.28125 21.34375 -23.375 C 20.425781 -24.476562 19.488281 -25.585938 18.53125 -26.703125 C 17.570312 -27.816406 16.632812 -28.9375 15.71875 -30.0625 C 14.800781 -31.1875 13.976562 -32.335938 13.25 -33.515625 C 12.53125 -34.703125 11.953125 -35.925781 11.515625 -37.1875 C 11.078125 -38.457031 10.859375 -39.796875 10.859375 -41.203125 C 10.859375 -43.097656 11.265625 -44.847656 12.078125 -46.453125 C 12.890625 -48.066406 14.015625 -49.457031 15.453125 -50.625 C 16.890625 -51.800781 18.59375 -52.722656 20.5625 -53.390625 C 22.53125 -54.054688 24.675781 -54.390625 27 -54.390625 C 28.945312 -54.390625 30.703125 -54.148438 32.265625 -53.671875 C 33.835938 -53.191406 35.179688 -52.507812 36.296875 -51.625 C 37.410156 -50.75 38.269531 -49.6875 38.875 -48.4375 C 39.488281 -47.195312 39.796875 -45.816406 39.796875 -44.296875 Z M 39.796875 -44.296875"
			},
			{
				"x":"41.027344",
				"y":"72",
				"d":"M 30.09375 -12.90625 C 29.082031 -11.21875 27.925781 -9.578125 26.625 -7.984375 C 25.332031 -6.390625 23.867188 -4.96875 22.234375 -3.71875 C 20.609375 -2.476562 18.804688 -1.476562 16.828125 -0.71875 C 14.847656 0.0390625 12.671875 0.421875 10.296875 0.421875 C 8.585938 0.421875 7.039062 0.207031 5.65625 -0.21875 C 4.28125 -0.65625 3.097656 -1.332031 2.109375 -2.25 C 1.128906 -3.164062 0.375 -4.34375 -0.15625 -5.78125 C -0.6875 -7.226562 -0.953125 -8.96875 -0.953125 -11 C -0.953125 -12.382812 -0.8125 -13.976562 -0.53125 -15.78125 C -0.25 -17.59375 0.191406 -19.445312 0.796875 -21.34375 C 1.410156 -23.238281 2.222656 -25.097656 3.234375 -26.921875 C 4.242188 -28.753906 5.460938 -30.390625 6.890625 -31.828125 C 8.316406 -33.273438 9.988281 -34.4375 11.90625 -35.3125 C 13.832031 -36.195312 16.03125 -36.640625 18.5 -36.640625 C 20.113281 -36.640625 21.429688 -36.441406 22.453125 -36.046875 C 23.472656 -35.660156 24.28125 -35.148438 24.875 -34.515625 C 25.46875 -33.890625 25.875 -33.160156 26.09375 -32.328125 C 26.320312 -31.492188 26.4375 -30.632812 26.4375 -29.75 C 26.4375 -27.65625 25.976562 -25.710938 25.0625 -23.921875 C 24.15625 -22.128906 22.929688 -20.566406 21.390625 -19.234375 C 19.859375 -17.898438 18.09375 -16.820312 16.09375 -16 C 14.101562 -15.175781 12.03125 -14.695312 9.875 -14.5625 C 9.832031 -14.019531 9.789062 -13.507812 9.75 -13.03125 C 9.71875 -12.550781 9.703125 -12.097656 9.703125 -11.671875 C 9.703125 -10.453125 9.8125 -9.445312 10.03125 -8.65625 C 10.257812 -7.875 10.585938 -7.257812 11.015625 -6.8125 C 11.453125 -6.375 11.992188 -6.066406 12.640625 -5.890625 C 13.285156 -5.710938 14.039062 -5.625 14.90625 -5.625 C 16.332031 -5.625 17.625 -5.8125 18.78125 -6.1875 C 19.945312 -6.5625 21.035156 -7.078125 22.046875 -7.734375 C 23.054688 -8.390625 24.007812 -9.160156 24.90625 -10.046875 C 25.8125 -10.941406 26.722656 -11.894531 27.640625 -12.90625 Z M 10.4375 -18 C 11.800781 -18.09375 13.113281 -18.476562 14.375 -19.15625 C 15.644531 -19.84375 16.757812 -20.726562 17.71875 -21.8125 C 18.675781 -22.90625 19.441406 -24.148438 20.015625 -25.546875 C 20.597656 -26.941406 20.890625 -28.390625 20.890625 -29.890625 C 20.890625 -30.773438 20.757812 -31.488281 20.5 -32.03125 C 20.238281 -32.570312 19.769531 -32.84375 19.09375 -32.84375 C 18.175781 -32.84375 17.273438 -32.398438 16.390625 -31.515625 C 15.515625 -30.640625 14.691406 -29.5 13.921875 -28.09375 C 13.148438 -26.6875 12.460938 -25.097656 11.859375 -23.328125 C 11.265625 -21.554688 10.789062 -19.78125 10.4375 -18 Z M 10.4375 -18"
			},
			{
				"x":"68.660156",
				"y":"72",
				"d":"M 22.25 -12.90625 C 21.570312 -10.957031 20.816406 -9.269531 19.984375 -7.84375 C 19.148438 -6.414062 18.28125 -5.207031 17.375 -4.21875 C 16.476562 -3.238281 15.554688 -2.441406 14.609375 -1.828125 C 13.660156 -1.222656 12.738281 -0.753906 11.84375 -0.421875 C 10.957031 -0.0859375 10.109375 0.132812 9.296875 0.25 C 8.492188 0.363281 7.773438 0.421875 7.140625 0.421875 C 6.054688 0.421875 5.046875 0.257812 4.109375 -0.0625 C 3.171875 -0.394531 2.359375 -0.910156 1.671875 -1.609375 C 0.984375 -2.316406 0.4375 -3.207031 0.03125 -4.28125 C -0.363281 -5.363281 -0.5625 -6.664062 -0.5625 -8.1875 C -0.5625 -8.84375 -0.515625 -9.5625 -0.421875 -10.34375 C -0.328125 -11.132812 -0.1875 -11.988281 0 -12.90625 L 4.328125 -33.125 L 1.9375 -33.125 L 2.53125 -36 L 4.890625 -36 L 7.0625 -45.953125 L 17.71875 -47.390625 L 15.265625 -36 L 19.578125 -36 L 19.015625 -33.125 L 14.703125 -33.125 L 10.09375 -11.53125 C 9.976562 -11.132812 9.898438 -10.742188 9.859375 -10.359375 C 9.828125 -9.972656 9.8125 -9.59375 9.8125 -9.21875 C 9.8125 -8.207031 10.046875 -7.507812 10.515625 -7.125 C 10.984375 -6.738281 11.722656 -6.546875 12.734375 -6.546875 C 13.429688 -6.546875 14.113281 -6.703125 14.78125 -7.015625 C 15.445312 -7.328125 16.0625 -7.765625 16.625 -8.328125 C 17.1875 -8.890625 17.691406 -9.5625 18.140625 -10.34375 C 18.585938 -11.132812 18.953125 -11.988281 19.234375 -12.90625 Z M 22.25 -12.90625"
			},
			{
				"x":"87.890625",
				"y":"72",
[... cut ...]
	for (var index = 0; index < paths.length; index++)
	{
		var d = paths[index]['d'];
		var x = paths[index]['x'];
		var y = paths[index]['y'];
		var name = 'glyph_' + index;
		var model = makerjs.importer.fromSVGPathData(d, importOptions);
		// Setting the origin below causes the script to hang
		model.origin = [x, y];
		scene.models[name] = model;
	}
	
	var exportAttributes = {
		'svgAttrs': {
			'xmlns':'http://www.w3.org/2000/svg'
		}
	};
	
	var svg = makerjs.exporter.toSVG(scene, exportAttributes);
```

adamdyson, 2017-06-26:
I've narrowed down the problem to how the SVG paths are being parsed. The paths in the code above are character glyphs. I've also try parsing the same paths using this library https://github.com/Pomax/js-svg-path which also fails to parse the paths at index 1,5,6,7.

```

var API = require('js-svg-path');

var data = {
	"paths":[
		{
			"x":"0",
			"y":"72",
			"d":"M 39.796875 -44.296875 C 39.796875 -43.285156 39.660156 -42.363281 39.390625 -41.53125 C 39.117188 -40.707031 38.75 -39.992188 38.28125 -39.390625 C 37.8125 -38.796875 37.238281 -38.332031 36.5625 -38 C 35.882812 -37.675781 35.144531 -37.515625 34.34375 -37.515625 C 33.4375 -37.515625 32.613281 -37.707031 31.875 -38.09375 C 31.132812 -38.476562 30.492188 -39.035156 29.953125 -39.765625 C 30.847656 -40.421875 31.5625 -41.289062 32.09375 -42.375 C 32.632812 -43.46875 32.90625 -44.632812 32.90625 -45.875 C 32.90625 -47.425781 32.445312 -48.648438 31.53125 -49.546875 C 30.625 -50.453125 29.375 -50.90625 27.78125 -50.90625 C 26.8125 -50.90625 25.921875 -50.722656 25.109375 -50.359375 C 24.304688 -49.992188 23.617188 -49.488281 23.046875 -48.84375 C 22.472656 -48.207031 22.023438 -47.441406 21.703125 -46.546875 C 21.390625 -45.660156 21.234375 -44.6875 21.234375 -43.625 C 21.234375 -42.03125 21.617188 -40.53125 22.390625 -39.125 C 23.171875 -37.71875 24.144531 -36.332031 25.3125 -34.96875 C 26.488281 -33.613281 27.753906 -32.238281 29.109375 -30.84375 C 30.472656 -29.457031 31.738281 -27.972656 32.90625 -26.390625 C 34.082031 -24.804688 35.054688 -23.09375 35.828125 -21.25 C 36.597656 -19.414062 36.984375 -17.382812 36.984375 -15.15625 C 36.984375 -12.789062 36.488281 -10.601562 35.5 -8.59375 C 34.519531 -6.59375 33.15625 -4.859375 31.40625 -3.390625 C 29.664062 -1.929688 27.617188 -0.785156 25.265625 0.046875 C 22.910156 0.878906 20.359375 1.296875 17.609375 1.296875 C 15.316406 1.296875 13.238281 0.992188 11.375 0.390625 C 9.507812 -0.203125 7.914062 -1.050781 6.59375 -2.15625 C 5.269531 -3.269531 4.242188 -4.597656 3.515625 -6.140625 C 2.785156 -7.691406 2.421875 -9.421875 2.421875 -11.328125 C 2.421875 -12.921875 2.675781 -14.40625 3.1875 -15.78125 C 3.707031 -17.164062 4.410156 -18.367188 5.296875 -19.390625 C 6.191406 -20.410156 7.25 -21.210938 8.46875 -21.796875 C 9.6875 -22.378906 10.988281 -22.671875 12.375 -22.671875 C 14.0625 -22.671875 15.476562 -22.25 16.625 -21.40625 C 17.78125 -20.5625 18.5 -19.40625 18.78125 -17.9375 C 17.507812 -17.9375 16.367188 -17.757812 15.359375 -17.40625 C 14.359375 -17.050781 13.503906 -16.554688 12.796875 -15.921875 C 12.097656 -15.296875 11.550781 -14.53125 11.15625 -13.625 C 10.769531 -12.71875 10.578125 -11.703125 10.578125 -10.578125 C 10.578125 -8.472656 11.191406 -6.804688 12.421875 -5.578125 C 13.660156 -4.347656 15.320312 -3.734375 17.40625 -3.734375 C 18.664062 -3.734375 19.832031 -3.957031 20.90625 -4.40625 C 21.988281 -4.863281 22.914062 -5.488281 23.6875 -6.28125 C 24.46875 -7.082031 25.078125 -8.035156 25.515625 -9.140625 C 25.960938 -10.242188 26.1875 -11.441406 26.1875 -12.734375 C 26.1875 -14.046875 25.96875 -15.300781 25.53125 -16.5 C 25.101562 -17.707031 24.523438 -18.878906 23.796875 -20.015625 C 23.078125 -21.160156 22.257812 -22.28125 21.34375 -23.375 C 20.425781 -24.476562 19.488281 -25.585938 18.53125 -26.703125 C 17.570312 -27.816406 16.632812 -28.9375 15.71875 -30.0625 C 14.800781 -31.1875 13.976562 -32.335938 13.25 -33.515625 C 12.53125 -34.703125 11.953125 -35.925781 11.515625 -37.1875 C 11.078125 -38.457031 10.859375 -39.796875 10.859375 -41.203125 C 10.859375 -43.097656 11.265625 -44.847656 12.078125 -46.453125 C 12.890625 -48.066406 14.015625 -49.457031 15.453125 -50.625 C 16.890625 -51.800781 18.59375 -52.722656 20.5625 -53.390625 C 22.53125 -54.054688 24.675781 -54.390625 27 -54.390625 C 28.945312 -54.390625 30.703125 -54.148438 32.265625 -53.671875 C 33.835938 -53.191406 35.179688 -52.507812 36.296875 -51.625 C 37.410156 -50.75 38.269531 -49.6875 38.875 -48.4375 C 39.488281 -47.195312 39.796875 -45.816406 39.796875 -44.296875 Z M 39.796875 -44.296875"
		},
		{
			"x":"41.027344",
			"y":"72",
			"d":"M 30.09375 -12.90625 C 29.082031 -11.21875 27.925781 -9.578125 26.625 -7.984375 C 25.332031 -6.390625 23.867188 -4.96875 22.234375 -3.71875 C 20.609375 -2.476562 18.804688 -1.476562 16.828125 -0.71875 C 14.847656 0.0390625 12.671875 0.421875 10.296875 0.421875 C 8.585938 0.421875 7.039062 0.207031 5.65625 -0.21875 C 4.28125 -0.65625 3.097656 -1.332031 2.109375 -2.25 C 1.128906 -3.164062 0.375 -4.34375 -0.15625 -5.78125 C -0.6875 -7.226562 -0.953125 -8.96875 -0.953125 -11 C -0.953125 -12.382812 -0.8125 -13.976562 -0.53125 -15.78125 C -0.25 -17.59375 0.191406 -19.445312 0.796875 -21.34375 C 1.410156 -23.238281 2.222656 -25.097656 3.234375 -26.921875 C 4.242188 -28.753906 5.460938 -30.390625 6.890625 -31.828125 C 8.316406 -33.273438 9.988281 -34.4375 11.90625 -35.3125 C 13.832031 -36.195312 16.03125 -36.640625 18.5 -36.640625 C 20.113281 -36.640625 21.429688 -36.441406 22.453125 -36.046875 C 23.472656 -35.660156 24.28125 -35.148438 24.875 -34.515625 C 25.46875 -33.890625 25.875 -33.160156 26.09375 -32.328125 C 26.320312 -31.492188 26.4375 -30.632812 26.4375 -29.75 C 26.4375 -27.65625 25.976562 -25.710938 25.0625 -23.921875 C 24.15625 -22.128906 22.929688 -20.566406 21.390625 -19.234375 C 19.859375 -17.898438 18.09375 -16.820312 16.09375 -16 C 14.101562 -15.175781 12.03125 -14.695312 9.875 -14.5625 C 9.832031 -14.019531 9.789062 -13.507812 9.75 -13.03125 C 9.71875 -12.550781  [... cut ...]  13.125 -24.867188 12.796875 -24.007812 12.515625 -23.09375 L 7.625 0 Z M -2.75 0"
		}
	]
};

var paths = data.paths;
var outline = new API.Outline();
var parser = new API.SVGParser(outline);

for (var index = 0; index < paths.length; index++)
{
	try
	{
		parser.parse(data.paths[index].d, parseFloat(data.paths[index].x), parseFloat(data.paths[index].y));
		console.log('<path d="' + outline.toSVG() + '" />');
	}
	catch (error)
	{
		console.log('Failed to parse glyph at index: ' + index);
	}
}
```

danmarshall, 2017-06-26:
Hi Adam, I'm away on vacation without my computer, I will look into this when I get back in a few days. Thanks! -Dan

danmarshall, 2017-06-26:
This may be related to a recent update. Can you try with the package a couple versions back?

adamdyson, 2017-06-27:
@danmarshall I'm not sure what version I was using or when I last updated my packages (apologies I should have checked), but I can confirm the example is now working with the latest version `0.9.58`. Enjoy your holiday.

closed 2017-06-27 by adamdyson