microsoft/maker.js #271: Build failing

rienafairefr, 2017-07-04:
I couldnt get maker.js to build, my environment is `node 6.11.0`, `npm 3.10.10`, `tsc 2.4.1`, I tried 

    npm install
    npm run build

but `npm run build` fails at `tsc -p target`, which in turn fails at compiling chain.ts and openjscad.ts: 

    > tsc -p target
    src/core/chain.ts(297,76): error TS2345: Argument of type 'true | IContainChainsOptions' is not assignable to parameter of type 'IContainChainsOptions'.
      Type 'true' has no properties in common with type 'IContainChainsOptions'.
    src/core/openjscad.ts(133,21): error TS2394: Overload signature is not compatible with function implementation.

I went back a few commits and the first error went away, but the second one didn't, so I guess something broke about openjscad in makerjs in recent versions ?

danmarshall, 2017-07-04:
Oh I'm, not using TypeScript 2.4 yet - perhaps the new compiler is more strict. I will give it a try.

rienafairefr, 2017-07-04:
OK, that wasthe problem. I see a soft dependency on typescript 2.1 in package.json, I installed that and now it compiles fine.

danmarshall, 2017-07-08:
Thanks for your help, hope it's working now 👍

closed 2017-07-08 by danmarshall