microsoft/maker.js #289: fill / stroke by layer?

forresto, 2017-08-05:
Is there a way to override svg fill/stroke per layer? I thought `layer.exporterOptions` would do that.

forresto, 2017-08-05:
Found this: https://github.com/Microsoft/maker.js/issues/248 ... Will play some more.

danmarshall, 2017-08-05:
Apologies, I need to update the SVG export documentation with details. For stroke only, you can name the layer with one of 17 simple colors:
http://microsoft.github.io/maker.js/docs/advanced-drawing/#Layers
For full control you can add a `layerOptions` property to your exortOptions object. The layerOptions should be a map, with keys being your layer names, and values being the options object for that layer. 
Each layer options object has these properties that you may set: `fill`, `stroke` (color), `strokeWidth`, and `cssStyle` (which can contain any CSS styling)

danmarshall, 2017-08-08:
Documentation updated: http://microsoft.github.io/maker.js/docs/exporting/#SVG

Next, I suppose that people may also want to specify DOM order in their output of layers. There will need to be a property added someplace. Either an `order` number on each layer option object, or a new property `layerOrder` which would be an array of layer names in order.

StollToaldo, 2017-08-16:
I'm trying to use `cssStyle` but I think I didin't understand how to use it. I'm trying to use like this:

`myLayerName: {
          strokeWidth: 3,
          stroke: '#9e9e9e',
          fill: 'black',
          cssStyle: {
            fillOpacity: 0.5
          }
        }`

Is that correct?

danmarshall, 2017-08-16:
@StollToaldo no - CSSStyle is a string, so:
```
cssStyle: "fillOpacity: 0.5; otherProperty:otherValue; "
```

StollToaldo, 2017-08-16:
Thanks @danmarshall, but it still not working. Here is my code:

```
const svg = makerjs.exporter.toSVG(svgModels, {
      useSvgPathOnly: false,
      svgAttrs: {
        width: '100%',
        height: '440px'
      },

      layerOptions: {

        safe: {
          strokeWidth: 5,
          stroke: '#9e9e9e',
        },

        notSafe: {
          strokeWidth: 5,
          stroke: '#e06666',
          fill: 'black',
          cssStyle: 'fillOpacity: 0.5; '
        }

      }
     })
```

Am I doing something wrong? `strokeWidth`, `stroke` and `fill` are working fine.

danmarshall, 2017-08-17:
@stolltoaldo my bad, there is a missing piece of code, I will release tonight. Your code is right. One last thing, after release, the style should be hyphenated instead of camelCase: 
"fill-opacity: 0.5"

danmarshall, 2017-08-17:
@StollToaldo 0.9.69 is published to npm. Can you try now?

StollToaldo, 2017-08-17:
@danmarshall , awesome, now it's working. I really appreciate your help.

closed 2017-08-17 by danmarshall