microsoft/maker.js #291: Questions: DXF Import

z3dev, 2017-08-11:
@danmarshall 

I'm working on the DXF import into OpenJSCAD.org, as we want to be able to import/export all supported formats.

It seems that you have browsed around Autocad far more than I have, and I have a few questions.
- Are there default color mappings? It seems that DXF just has a number, but it would be nice to map this to something useful.
- Is there a variable that contains the 'unit' for a drawing? I will default to milimeter but it certainly would be nice to know how the units are passed between applications.

Any other tips?

danmarshall, 2017-08-11:
Hi @z3dev :-)
I've used [this color mapping](http://sub-atomic.com/~moses/acadcolors.html) which casts to hex and rgb. I've mapped 17 of these to the primitive named web colors [here](https://github.com/Microsoft/maker.js/blob/master/src/core/exporter.ts#L33).

There is a header named `$INSUNITS` that sets the unit type, which I've enumerated a few [here](https://github.com/Microsoft/maker.js/blob/master/src/core/dxf.ts#L217).

I've seen some DXF parsers on [NPM](https://www.npmjs.com/search?q=dxf&page=1&ranking=optimal) . [This one](https://www.npmjs.com/package/dxf) and [this one](https://www.npmjs.com/package/dxf-parser) seem pretty good though I haven't used them personally.

z3dev, 2017-08-11:
@danmarshall Thanks

I'll use the same color mapping, so we stay in sync. ;)

I didn't like any of the DXF parsers as most focused on 2D spaces and full documents. I'm really interested in 3D models, and once satisfied will add support for 2D as well.

I'm trying to write the parser in a very simple way, allowing others to use the parser to obtain the information required, nothing more. DXF files are packed full of useless information, and really only useful in Autocad applications.

I'll let you know when finished. You may be able to add support for DXF fairly easy.

danmarshall, 2017-08-11:
@z3dev great, keep me updated!

rkrishnasanka, 2018-08-27:
@danmarshall I've been using [https://www.npmjs.com/package/dxf-parser](https://www.npmjs.com/package/dxf-parser) in my project, I think its good because it generates an object representation that you can use however you want at the end of the day.

danmarshall, 2018-08-27:
Thanks for the tip @rkrishnasanka !

rkrishnasanka, 2018-08-29:
@danmarshall no issues. Here's the project I'm working on http://3duf.org/. I came across your project when I was looking for ways to export DXF.

danmarshall, 2019-08-09:
I'm going to close this issue since DXF import will probably be a separate lib.

closed 2019-08-09 by danmarshall