microsoft/maker.js #300: Use of updated OpenJscad & CSG

kaosat-dev, 2017-09-22:
Hi folks ! 
Great project :)
I am one of the main maintainers of OpenJscad and wanted to let you know there have been major updates to the project:
- it is now a github org https://github.com/jscad
- multiple related projects have been transered to the org : openscad translator csg.js etc
- code base has been cleaned up (moved partially to es6 etc), tests added etc
- code base is now modular, and all sub packages are now [also on npm](https://www.npmjs.com/org/jscad)

It would be great if you could switch to the more up to date packages , as they are more atomic and much more maintained.
If you need any help, or information, or have feedback please let us know !
Cheers.

z3dev, 2017-09-22:
And thanks for using the libraries!

danmarshall, 2017-09-22:
@kaosat-dev @z3dev thank you and likewise :) 
Great work on your re-org's. Glad to see the updates. I will update and let you know how it goes.
Cheers!

danmarshall, 2017-11-27:
I've been looking at OpenJSCAD in depth this weekend. 
I can begin the update when [the PR for status updates](https://github.com/jscad/io/pull/49) gets merged.

danmarshall, 2017-12-09:
It's in! I'll have a nice demo soon :)

kaosat-dev, 2017-12-11:
woohoo, well done :) Please let us know when the demo is ready ;)

closed 2017-12-21 by danmarshall

danmarshall, 2017-12-21:
This turned out to be a pretty big change! But I'm happy with the way it turned out. I've been wanting to update this interface for quite some time. I've deprecated the previous interfaces:
* https://maker.js.org/docs/api/modules/makerjs.exporter.html#toopenjscad
* https://maker.js.org/docs/api/modules/makerjs.exporter.html#tostl
They still work, but do it "the old way" of expecting the globals CAG & CSG.

The new interfaces are: 
* https://maker.js.org/docs/api/modules/makerjs.exporter.html#tojscadcag
* https://maker.js.org/docs/api/modules/makerjs.exporter.html#tojscadcsg
* https://maker.js.org/docs/api/modules/makerjs.exporter.html#tojscadstl

Best demo you can see is the new "options panel" in [the playground](https://maker.js.org/playground/?script=exoPanel), after you click the STL export button:
[image]

The progress bar uses the new "statusCallback" in the serializer :)

PR #306 (2017-12-21): new build 0.9.77
Fixes #300