microsoft/maker.js #307: Model mirroring does not mirror layer colors

berrydan, 2017-12-23:
A model with multiple layers will successfully rotate and retain its layer attribute, but mirroring a model will drop the layer attribute of all mirrored objects.

    var makerjs = require('makerjs');
    
    var thingy = {
      models: {
        "yellow": {
          layer: "yellow",
          paths: {
            "layer1": new makerjs.paths.Line([0,0],[10,0])
          }
        },
        "red": {
          layer: "red",
          models: {
            "layer2": new makerjs.models.ConnectTheDots(false, [ [0,0],[0,5],[5,10],[10,10],[10,0] ])
          }
        }
      }
    }
      
    var thingyRotated = makerjs.cloneObject( thingy );
    makerjs.model.rotate( thingyRotated, 90, [-1,0] );
      
    var thingyMirrored = makerjs.model.mirror( thingyRotated, false, true );
    
    var myModel = {
      "Right": thingy,
      "TopLeft": thingyRotated,
      "BottomLeft": thingyMirrored
    }
    
    //...Uncomment to 'fix' the broken mirror
    //myModel.BottomLeft.models["yellow"].layer = "yellow";
    //myModel.BottomLeft.models["red"].layer = "red";
    
    this.models = myModel;

danmarshall, 2017-12-23:
Good catch! Fix coming right up :)

danmarshall, 2017-12-23:
Should now be fixed, in version 0.9.78 - can you try again?

berrydan, 2017-12-24:
Exactly what I did for testing. ;-) Cheers

closed 2017-12-24 by berrydan