microsoft/maker.js #314: DXF Polylines

danmarshall, 2018-01-02:
The DXF export currently only does individual entities. Could add an option to detect chains and emit Polylines.

z3dev, 2018-01-03:
FYI, we are moving towards arrays for both input to export functions, and arrays as output from import functions at JSCAD. This will take a little time to get right but till make models with 2D and 3D objects possible, as well as models with more than one object. Should be easier to understand as well.

zrichmond, 2018-01-03:
Supporting polylines for the dxf export would definitely be helpful.  In addition to @mdales situation it would also allow for some CAM software when generating gcode for a router/milling machine to be able to cut contours that allow for tool radius offsetting based on being inside or outside of the polyline, and being able to determine cut direction (CW/CCW)  all of which matter when it comes to routing/milling cutting techniques.

z3dev, 2018-01-03:
@zrichmond Could you add a few thoughts to this issue for JSCAD as well? I’m about to rewrite the export for DXF.

danmarshall, 2018-01-07:
After doing a  little research, I've found:
* [LWPOLYLINE](https://www.autodesk.com/techpubs/autocad/acad2000/dxf/lwpolyline_dxf_06.htm) is only 2D, supports line segments and circular arcs.
* [POLYLINE](https://www.autodesk.com/techpubs/autocad/acad2000/dxf/polyline_dxf_06.htm) can be 3D, supports line segments and higher order (but not circular?) curves.

So it looks like Maker.js aligns well with LWPOLYLINE. @zrichmond @mdales let me know if LWPOLYLINE will work for your scenarios.

@z3dev it appears the POLYLINE may be the better fit for JSCAD depending on the scenario. If you are outputting 2D then you may also look at LWPOLYLINE.

z3dev, 2018-01-07:
@danmarshall thanks for the information.

I’m planing on using LWPOLYLINE for 2D paths. Sadly, JSCAD doesn’t have the proper lines or line segments yet, which would be best as POLYLINE.

For 3D, I’m planning on using 3DFACE which is more efficient that other DXF entities.

danmarshall, 2018-01-08:
(These are just my collection of LWPOLYLINE implementation / DXF file format footnotes from 3 hours of research)
* A formula for [calculating bulge of arcs in a LWPOLYLINE](http://lee-mac.com/bulgeconversion.html#arcbulge)
* LWPOLYLINE is an entity [introduced after DXF R12](https://forums.autodesk.com/t5/net/dxf-format/td-p/2174884).
* DXF may have an [$ACADVER](http://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-A85E8E67-27CD-4C59-BE61-4DC9FADBE74A) in the header [corresponding to the revision number](https://forums.autodesk.com/t5/autocad-forum/how-to-find-version-what-version-dxf-format-we-are-saving-the/td-p/2442363)
* DXF > R12 must have [this structure](http://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-D939EA11-0CEC-4636-91A8-756640A031D3) and [these tables](http://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-A9FD9590-C97B-4E41-9F26-BD82C34A4F9F) - if it does not have these, [it may error out](https://forums.autodesk.com/t5/autocad-forum/unable-to-open-dxf-file-in-autocad/td-p/5652821). In my testing, most third party DXF viewers will actually render the LWPOLYLINE as a normal entity - but official AutoCAD programs will not without these tables.
* Each table has [this format](http://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-5AB9300F-F0AC-4ADE-89EA-A9D1D152D8B8)
* The above implies an incremental handle, which the max should be recorded in [$HANDSEED](http://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-A85E8E67-27CD-4C59-BE61-4DC9FADBE74A)

Also, it appears that POLYLINE (not LW-) *is* part of the earlier R12 version, which may avoid having to add all these extra tables. In the case of Maker.js, most of the tables are just no-ops, other than the Layers table. So, I may end up backtracking and go with POLYLINE, if I can find a way to make it work for circular arcs.

mdales, 2018-01-08:
Apologies @danmarshall, I've been away form the keyboard for a few days.

So, before I wrote the generaor, I was working off a design that I tweaked and then exported to DXF from Cut 2D software. If I look inside that design, then it is using POLYLINE, not LWPOLYLINE. I guess it could have been exported as R12, but Cut 2D didn't give me any options, I just went with the default. 

I also know that in general I find the laser cutter tool I use seems to barf on later versions than R12, so if that's easier for you, then that seems to be a good option from my personal use case, though I appreciate not suitable for a broader audience perhaps.

I've put that original DXF file here incase it's of use as reference: https://gist.github.com/mdales/a2c84c3d4e640fb903b6fc4b7389e54e

My DXF knowledge is pretty minimal, my original request about POLYLINEs was based on comparing the output of maker.js with that DXF.

z3dev, 2018-01-08:
LWPOLYLINE and POLYLINE have line segments with BLUG (arc segments). For POLYLINE, the BULG is added to each VERTEX.

closed 2018-01-10 by danmarshall

danmarshall, 2018-01-10:
@z3dev thanks!

danmarshall, 2018-01-10:
@mdales Thanks for the feedback. I have just published a new version 0.9.82 with DXF Polyline feature. However, it seems to only work in some DXF viewers. Not all I have tested will open the file :(
Can you go into a [model in the playground](https://maker.js.org/playground/?script=Square) and then:
* Click the `DXF` download button
* Check the `Use POLYLINE` checkbox:
[image]
* Click the `Generate` button
* Click the `Download .DXF` button
* Open the DXF file in the app(s) you use
* Verify that you can see the polyline, and that it is one entity

When you want to change your app to use POLYLINE, just add an [options object](https://maker.js.org/docs/api/interfaces/makerjs.exporter.idxfrenderoptions.html#usepolyline) like this to `makerjs.exporter.toDXF(model, { usePOLYLINE: true })`

PR #317 (2018-01-10): new build 0.9.82
Fixes #314