microsoft/maker.js #318: Best practices for unioning models

davidisaaclee, 2018-01-16:
I'm using this library to create a model from a bunch of SVG paths, and then export the model as a DXF for engraving. I want the model to be "filled" (so as to match the look of the filled, closed SVG paths). The engraving software I'm using has a way of filling in paths, but does so in an "even-odd" way, when what I want is a "non-zero" fill.

[image]

It seems like what I want to do is union all of my input paths together, and export the unioned model to DXF. So I'm taking each model that I get from the SVG importer, and reducing them into a single unioned model:
```javascript
function unionModels(models) {
  return models
    .reduce(
      (acc, model, index) =>
        makerjs.$(model)
          .combineUnion(acc)
          .originate() // not certain that this is helpful
          .$result,
      {});
}
```

However, each step of the reduction seems to take longer and longer, and also seems to consume more and more memory. In my use case, there can be a lot of paths as inputs, and these paths can be pretty complex (including all different kinds of SVG path commands, up to 100kb SVG file size).

**What are the best practices for combining many models via `combineUnion`? Is there any way I can make this faster or use less memory?**

Alternatively, is there another solution to my use case? (Unfortunately, changing my engraving software to something that supports more advanced fill methods is not an option...)

Thanks in advance!

danmarshall, 2018-01-16:
Hi @davidisaaclee, creating a faster algorithm for combining is high on my list. One of my thoughts for a future API is just passing an array of models so that we don't need to keep an accumulator, as in your example (and in the [expandPaths code](https://github.com/Microsoft/maker.js/blob/master/src/core/expand.ts)).

 Until then you can hopefully shave a little time by caching measurements. This means that you can compare bounding boxes instead of computing path intersections.

Outside of your loop, create an options object to do the caching:
```
combineOptions = {};
```

You'll need to use the longhand form of combine, instead of the shorthand `combineUnion`, so:
```
combine(modelA, modelB, false, true, false, true, combineOptions);
```
Using modelA as the accumulator, means we want to keep the measurements of modelA. Since modelB is different each iteration, we will delete the measurements of modelB. See [this line](https://github.com/Microsoft/maker.js/blob/master/src/core/expand.ts#L132) for an example.

This might not radically improve your situation. Other than this, I would suggest parallelizing where possible (unioning groups of models which you know do not overlap other groups), and caching results of unions, so you aren't running the same computation again.

danmarshall, 2018-01-16:
Also see the [ICombineOptions interface](https://maker.js.org/docs/api/interfaces/makerjs.icombineoptions.html).

danmarshall, 2018-01-16:
Oh - and I don't think you need to `.originate()` in your loop ;-)

davidisaaclee, 2018-01-16:
Thank you for the advice – I'll give this a shot.

closed 2018-02-09 by davidisaaclee

danmarshall, 2018-02-09:
BTW I'm working on some new code for this, I hope to have it out soon 🙂