microsoft/maker.js #327: Feature Request: Scaling Models by Axis

blainedwards8, 2018-03-05:
This library is awesome and I have been waiting for something like this for a long time.  I have bee playing around with library and noticed that a scaling feature is available but it is a uniform features (i.e. applied across both the X, and Y).  I am dealing with some issues with Text blocks that need to fit within a given set of dimensions.  While text, allows you to set a given size, it doesn't allow to manipulate the width for a given height.  That would be a very useful feature.

I have looked through the API and noticed a distort function for both the Path and Point modules.  It seems to have the ability to distort in a given direction.  I haven't had much luck using it.  However, I am not sure whether this would be the place to enter functionality.

I forked the project and am currently trying to update the scale function in the model, path, and point modules to include a feature allowing scaling in a given axis.  

I would be happy to contribute where I can to help make this an even more awesome project.  Thanks again.

danmarshall, 2018-03-05:
Hi Blaine, thanks for the feedback! 🥇 You are right - there is not yet a `model.distort(x, y)` function! It is on my [roadmap](https://github.com/Microsoft/maker.js/projects/3).
It would be great to accept your code contribution! If you know TypeScript, that is great - otherwise, this is your chance to learn it.
To implement the feature, all you'd need to do is create a function that:
* will return a new model (does not distort "in place")
* walk all paths of the model
* distort each path (which returns a model), and add this to the result

Hopefully [the source](https://github.com/Microsoft/maker.js/blob/master/src/core/model.ts) will provide some examples how to accomplish this. And if you can also write a unit test for your feature, that would be awesome too. 
Thanks!

blainedwards8, 2018-03-06:
Dan,

Thanks for the response.  I would be happy to help out with this feature.   After seeing this  projects, I began looking at TypeScript and am actually doing a project at work with it.  I should  be able to manage.  I will work out a solution and send it back.

Blaine

danmarshall, 2018-03-06:
Thanks Blaine! Please feel free to ask any questions along the way, it may save you some time.

One thing I will say, if you are using VSCode: I have noticed that it will sometimes automatically add `import` statements to the beginning of a file, when you type the name of a reference from another file. This is not desired, and will break the build. This project is not yet using the new module style. So if these appear, you can just delete them. The first time it happened to me, it took me an hour to figure out why.

danmarshall, 2018-08-13:
Hi @blainedwards8 , I finally got around to implementing this. Take a look at https://maker.js.org/docs/intermediate-drawing/#Distorting 👍

closed 2018-08-13 by danmarshall

joelabair, 2019-02-17:
@danmarshall i'm getting ```Uncaught TypeError: makerjs.model.distort is not a function``` w/ maker.js-0.9.17 's target/js/browser.maker.js

joelabair, 2019-02-17:
I see, the issue stems from npm's latest version is 0.13.1 and the github project's releases latest is 0.9.17

Obviously, the issue is not there in 0.13.1

sorry for the alarm

PR #356 (2018-08-13): model.distort
Fixes #327