microsoft/maker.js #328: Split arc into segments for DXF export

3kmch, 2018-04-06:
Hi

I recently updated my kerf reference generator script:
https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/0d7bbf16b193e010f20fee8230c8dd4c/raw/576664ffddf199f9f356ac6bd5cc6f74559f565c/kerf.js

It works perfectly as long as I don't import it into the crappy piece of software which drives the Chinese lasercutter. All the fillets are shown as chamfer.

[image]

Is there an easy method of splitting all arcs into segments or a path of segments? I guess this way the lasercutter would have less problems to get them right.   

It's bit embarrassing but I'm not able to get rid of the "Font Selector" in the customizer. How can I hardcode that? I'm not a coder. So please don't judge me by my second question and my code structure. ;-)

Thanks, Kurt

danmarshall, 2018-04-06:
Hi Kurt!
Thanks for posting this question, and we welcome coders of all levels 👍 

Here's a function that will straighten out those `RoundRectangle`s: 
```js
function straightenRoundRect(roundRect, facetSize) {

    //replace an arc (path) with a series of lines (model)
    function corner(id) {

        //get the arc and remove it from the roundRect
        var arc = roundRect.paths[id];
        delete roundRect.paths[id];

        //get the points along the arc, of the facetSize
        var points = makerjs.path.toKeyPoints(arc, facetSize);

        //connect the points as lines, add this model to roundRect
        var straightened = new makerjs.models.ConnectTheDots(false, points);
        roundRect.models[id] = straightened;
    }

    //create a place to put all the models
    roundRect.models = {};

    ["TopRight", "TopLeft", "BottomRight", "BottomLeft"].forEach(corner);
}
```

Now, about removing that font selector: You'll need to write some asynchronous code, since we have to wait for the font to download. If you are writing a custom web page, take a look at [this example](https://maker.js.org/docs/advanced-drawing/#Fonts%20and%20text) in the docs. In the Maker.js playground, there's a special `playgroundRender` function you can call once your asynchronous calls are complete and you'd like to render a model:
```js
//load a font asynchronously
opentype.load('/fonts/allertastencil/AllertaStencil-Regular.ttf', function (err, font) {

    if (err) {
        alert('the font could not be loaded :(');
    } else {
        //you can now pass the font to a function.

        //we will just create a Text model as an example:
        var textModel = new makerjs.models.Text(font, 'Hello', 100);

        //now call playgroundRender, which only works in the Maker.js playground app.
        playgroundRender(textModel);
    }
});
```

Finally - this is a really cool model! Would you mind sharing it as a demo that I can feature on the site?
-Dan

3kmch, 2018-04-07:
Hi Dan

Thank you for your reply and the detailed comments. I would love to see the model on your site. :)

The arcs are now perfect!
https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/0d7bbf16b193e010f20fee8230c8dd4c/raw/25123b27809f08b37b802d5604c7f5301747c4f7/kerf.js

Regarding the font stuff I have no clue how I could do that. Do I have to pack everything into this function?

Kurt

danmarshall, 2018-04-10:
Hi Kurt,
Yes you'd need to call you code from the function. If it's not a problem, can we just leave the font as  it is, with the Font selector?

3kmch, 2018-04-12:
Hi Dan

Now the script is finished and some documentation done on: https://www.thingiverse.com/thing:2854454

Thank you for your support. :)

Kurt

z3dev, 2018-04-12:
Cool design!

danmarshall, 2018-04-18:
Hi Kurt, your design is now featured on the site! 🥇 
You can update your URLs to https://maker.js.org/playground/?script=kerf-compensation and you won't need to see the warning message.
Thanks!
Dan

closed 2018-04-18 by danmarshall