microsoft/maker.js #329: Convert svg to dxf

usama-ayub, 2018-04-16:
Hi,

Anyone help how to convert svg to dxf 
it's urgent

danmarshall, 2018-04-16:
Hi Usama, can you tell me what types of SVG elements are in your document? Maker.js deals primarily with path data. See https://maker.js.org/docs/importing/#content for import techniques. You would need to traverse your SVG document and import each element accordingly.

usama-ayub, 2018-04-16:
Hi @danmarshall  thanks for your response i am use like this
maker.exporter.toDXF('./a.svg',{usePOLYLINE:true,units:'dxf'})

danmarshall, 2018-04-16:
Hi Usama, do you need to do this for only one SVG document, or do you need a script that works for a multitude of documents?

If you only need one document, I would simply use [Inkscape](https://inkscape.org/en/), which has a GUI that can export to DXF for you. 

If you are looking at creating a script that will process many SVG documents, you will need more consideration of the system you need to build.

Maker.js is a low level library that only works with [model objects](https://maker.js.org/docs/basic-drawing/#Models). You can create model objects by [importing SVG path data or points](https://maker.js.org/docs/importing/#content). Once you have a model, you can then [export the model to DXF](https://maker.js.org/docs/exporting/#DXF).

Maker.js will run in both Node.js and web browsers, each of these might have their own way of handling SVG files. So it would also help to know your environment.

danmarshall, 2018-04-17:
Hi Usama, how many documents do you need to convert?

usama-ayub, 2018-04-17:
only one

danmarshall, 2018-04-17:
For just one document, it will probably be easiest if you install Inkscape. It is not a library, it is a desktop app that you can install on Windows or Mac. No coding required, it's a GUI app. It should work for you, it is an extremely popular app! 😃👍

usama-ayub, 2018-04-18:
I have svg like this 
<svg width="100" height="100">
 <circle cx="50" cy="50" r="40" stroke="green" stroke-width="4" fill="yellow" />
</svg>
and i want to convert this into dxf

danmarshall, 2018-04-18:
Are you in Node.js or a browser environment?

danmarshall, 2018-04-19:
Here is a basic minimum example:

```html
<svg width="100" height="100">
    <circle cx="50" cy="50" r="40" stroke="green" stroke-width="4" fill="yellow" />
</svg>

<div id="dxf"></div>

<script>
    var model = {};
    var svg = document.getElementsByTagName('svg')[0];
    for (var i = 0; i < svg.children.length; i++) {
        var child = svg.children[i];
        switch (child.tagName) {
            case "circle":
                var origin = [+child.getAttribute('cx'), +child.getAttribute('cy')];
                var radius = +child.getAttribute('r');
                var circle = new makerjs.paths.Circle(origin, radius);
                circle.layer = child.getAttribute('stroke');
                makerjs.model.addPath(model, circle);
                break;
        }
    }
    document.getElementById('dxf').innerText = makerjs.exporter.toDXF(model);
</script>
```
From here you will have to make some choices, based on your requirements. There's a few things to consider:

* DXF does not have "fill" - so the `yellow` data is lost.
* `stroke-width` is not supported in Maker.js. Do you want a [ring](https://maker.js.org/playground/?script=Ring#params=[42,38]) instead of a circle?

usama-ayub, 2018-04-19:
no need this svg just example 
but svg look like this
[image]
[image]
[image]

danmarshall, 2018-04-19:
Remember that Maker.js is primarily line drawing. So you will not be able to convert every type of SVG element, such as images. Also remember that text is also treated as outlines in Maker.js, and that you will need to download a font - see the [documentation on fonts and text](https://maker.js.org/docs/advanced-drawing/#Fonts%20and%20text).

Pretty much the elements you'll need to add code for are `circle`, `ellipse`, `rect`, `polygon`, and `path`. See the documentation on [importing](https://maker.js.org/docs/importing/#content), noting that the y-coordinates in SVG are opposite in Maker.js.

Finally, you'll need to consider if you want to support SVG's `transform` attribute, which introduces additional math on each type of shape.

danmarshall, 2018-08-19:
Closing this issue for now, please reopen if there is anything else.

closed 2018-08-19 by danmarshall