microsoft/maker.js #332: Boolean subtraction fails for unknown reasons

yowzadave, 2018-05-04:
I'm doing a boolean operation on some models that have floating-point geometry (using mouse input to locate vertices). In nearly all cases, these work, but I've noticed that they occasionally fail, even when the models are overlapping and don't have coincident edges or anything obvious that should result in a failed operation. An example of one of these operations that fails is below. Any ideas how I might avoid these situations?

Thanks

```javascript
let combined = {
  models: {
    model1: {
      "paths": {
        "edge_0": {
          "type": "line",
          "origin": [0, 0],
          "end": [100, 0]
        },
        "edge_1": {
          "type": "line",
          "origin": [100, 0],
          "end": [100, 78.57143211364746]
        },
        "edge_2": {
          "type": "line",
          "origin": [100, 78.57143211364746],
          "end": [0, 100]
        },
        "edge_3": {
          "type": "line",
          "origin": [0, 100],
          "end": [0, 0]
        }
      }
    },
    model2: {
      "paths": {
        "corner": {
          "radius": 12,
          "origin": [88, 68.87044229483664],
          "startAngle": 0,
          "endAngle": 77.90524486342795,
          "type": "arc"
        },
        "edge1": {
          "type": "line",
          "origin": [100, 68.87044229483664],
          "end": [103, 68.87044229483664]
        },
        "edge2": {
          "type": "line",
          "origin": [103, 68.87044229483664],
          "end": [103.88833038657913, 81.71481990467888]
        },
        "edge3": {
          "type": "line",
          "origin": [103.88833038657913, 81.71481990467888],
          "end": [91.14293586684671, 83.53747865835727]
        },
        "edge4": {
          "type": "line",
          "origin": [91.14293586684671, 83.53747865835727],
          "end": [90.5143487, 80.60407139483664]
        }
      }
    }
  }
}

makerjs.model.combineSubtraction(combined.models.model1, combined.models.model2)
```

danmarshall, 2018-05-07:
Hi @yowzadave, it may be that some [rounding in the intersection code](https://github.com/Microsoft/maker.js/blob/master/src/core/intersect.ts#L318) may be the culprit. I will do some testing to see if removing it helps your case, and yet continues working for other test cases.

yowzadave, 2018-05-07:
Sounds good—would be great if that fixes the issue.

yowzadave, 2018-05-08:
@danmarshall yes, although that may not make the most sense in my case, since the model I'm subtracting is not always a rounded corner. If it happens that there is something particular about a rounded corner that makes it a bad candidate for a boolean operation, then I may handle it as a fillet in those scenarios. Thank you!

danmarshall, 2018-05-09:
This is a really good use case. I'll need to fortify the combine() algorithm to break paths where endpoints are extremely close to another path.

UXDart, 2018-05-18:
Hi! I have a similar problem I think, I'm new to MakerJS... I'm doing this:
I have two `models: { a: rect1, b: rect2 }`... 

rect1 = makerjs.importer.fromSVGPathData(path1)
with path1 = "M 8 37 L 53 37 A 8 8 0 0 0 61 29 L 61 8 A 8 8 0 0 0 53 0 L 8 0 A 8 8 0 0 0 0 8 L 0 29 A 8 8 0 0 0 8 37 Z"

rect2 path is "M 40 56 L 112 56 A 8 8 0 0 0 120 48 L 120 31 A 8 8 0 0 0 112 23 L 40 23 A 8 8 0 0 0 32 31 L 32 48 A 8 8 0 0 0 40 56 Z"

basically they are two rectangles but with rounded borders, one on topLeft, one on bottomRight, they cross in the middle

if I remove the rounded borders then, it seems to work the:
`let exporterPath = makerjs.exporter.toSVGPathData(mm, { fillRule: 'nonzero', });`

but it does something strange, nonzero vs evenodd does the opposite. if I add back the rounded borders they show the middle rectangle all the time. without rounded borders with nonzero it shows an empty rectangle in the middle... but IMO that should be with evenodd...

I'm doing something wrong?

Thanks!

UXDart, 2018-05-18:
I should really create a new issue I think... I just created a fiddler and I see is not really combining the two paths... how do I combine them using a fill method? ... I assumed that the exporter will combine them. I can only combine them using a type of union but not a evenodd method, so the rectangle in the middle will be empty... how I do that? TIA

danmarshall, 2018-05-18:
@UXDart I'm not sure what result you are trying to achieve. Let's start with some example code, and we will proceed from there. Also, perhaps you can provide a screenshot of your desired result.

here's some code you can copy and paste into [the playground](https://maker.js.org/playground/):
```js
var makerjs = require('makerjs');

var rect1 = makerjs.importer.fromSVGPathData("M 8 37 L 53 37 A 8 8 0 0 0 61 29 L 61 8 A 8 8 0 0 0 53 0 L 8 0 A 8 8 0 0 0 0 8 L 0 29 A 8 8 0 0 0 8 37 Z");
var rect2 = makerjs.importer.fromSVGPathData("M 40 56 L 112 56 A 8 8 0 0 0 120 48 L 120 31 A 8 8 0 0 0 112 23 L 40 23 A 8 8 0 0 0 32 31 L 32 48 A 8 8 0 0 0 40 56 Z");

var mm = {
  models: {
    rect1,
    rect2
  }
};

//render in playground
this.models = mm.models;

//makerjs.model.combineUnion(rect1, rect2);

let exporterPath = makerjs.exporter.toSVGPathData(mm, { fillRule: 'nonzero' });

console.log(exporterPath);
```

danmarshall, 2018-05-21:
@yowzadave - this issue should now be fixed, please give it a try. Thanks for reporting!
@UXDart - please feel free to open a new issue, unless you think this one needs to be re-opened.
Thanks,
Dan

yowzadave, 2018-05-21:
Hi @danmarshall, thanks for taking a look. I'm still seeing failed unions in some cases. Here's another example:

```javascript
let combined = {
  "models": {
    "composite": {
      "models": {
        "composite": {
          "paths": {
            "edge_0": {
              "type": "line",
              "origin": [
                0,
                16.428573608398438
              ],
              "end": [
                86.05464870576407,
                2.291022302323295
              ]
            },
            "edge_1": {
              "type": "line",
              "origin": [
                100.00000000576406,
                14.132289502323296
              ],
              "end": [
                100,
                100
              ]
            },
            "edge_2": {
              "type": "line",
              "origin": [
                100,
                100
              ],
              "end": [
                0,
                100
              ]
            },
            "edge_3": {
              "type": "line",
              "origin": [
                0,
                100
              ],
              "end": [
                0,
                16.428573608398438
              ]
            }
          }
        },
        "corner": {
          "paths": {
            "corner": {
              "radius": 12,
              "origin": [
                88.00000000576406,
                14.132289502323296
              ],
              "startAngle": 260.67045913231436,
              "endAngle": 359.99999988714217,
              "type": "arc"
            }
          }
        }
      }
    },
    "corner": {
      "paths": {
        "chamfer": {
          "type": "line",
          "origin": [
            100,
            90
          ],
          "end": [
            90,
            100
          ]
        },
        "edge1": {
          "type": "line",
          "origin": [
            100,
            90
          ],
          "end": [
            103,
            90
          ]
        },
        "edge2": {
          "type": "line",
          "origin": [
            103,
            90
          ],
          "end": [
            103.53553390593274,
            103.53553390593274
          ]
        },
        "edge3": {
          "type": "line",
          "origin": [
            103.53553390593274,
            103.53553390593274
          ],
          "end": [
            90,
            103
          ]
        },
        "edge4": {
          "type": "line",
          "origin": [
            90,
            103
          ],
          "end": [
            90,
            100
          ]
        }
      }
    }
  }
}

makerjs.model.combineSubtraction(combined.models.composite, combined.models.corner)
```

Thanks for your help!

danmarshall, 2018-05-23:
Hi @yowzadave - I took a look at this case.  We may be playing cat and mouse with floating point accuracy. So before I publish a minor fix, can you test as much as you can in your local copy:

Please change [this line](https://github.com/Microsoft/maker.js/blob/master/index.js#L4082) (if you are in Node.js) or [this line](https://github.com/Microsoft/maker.js/blob/master/target/js/browser.maker.js#L4289) (if you are in the browser) and remove one of the zeros. If that's not enough, remove one more zero.

Generally speaking, I think you could also use [makerjs.point.rounded](https://maker.js.org/docs/api/modules/makerjs.point.html#rounded) to round off some of your points. Doing that also made your example work. It will clean up some of the values such as `100.00000000576406` and `88.00000000576406`.

danmarshall, 2018-08-12:
@yowzadave seems to be working now, please feel free to re-open this issue if you see it again. Thanks

PR #335 (2018-05-21): new build 0.9.91
Fixes #332
closed 2018-08-12 by danmarshall