microsoft/maker.js #336: Creating dogbone cutouts for CNC

3kmch, 2018-05-22:
Some questions before I jump more into this topic:

A) Is there an easy way to load SVG-Files?
B) Can maker.js detect nested geometry?

My example below is not super usefuI. ;-) But the "engraved house" shows that some elements should not be touched. 

[image]

The final "Dogbone Converter" should work as follows:
1) Upload a SVG
2) Check the bounding box and scale up or down if needed
3) Define the dogbone radius
4) Download the SVG
5) Feed it into the CAM: In my case Laserweb4, which lacks of the dogbone function.

Thank you for any help :)
Kurt

Just for the record: my first tests
https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/1ac0b01d03f35499f2c806701bc7fb8e/raw/d849bb08359a5b2be474de6ebedfb13e8adf7776/dogbone.js

danmarshall, 2018-05-22:
Hi Kurt,
Although Maker.js exports SVG documents, there isn't a way to import them. This is because we can't make any assumptions about how the document is retrieved, and then how it is parsed or accessed.
For example, in the browser you might load the SVG on the network - or you might already have it available in the DOM. In Node.js, you might also load it from the network, or even from the file system, but the next step is to parse it - and there are [so many libraries for that](https://www.npmjs.com/search?q=xml).

Sounds like you might be making a browser app, and using local files. If so, I would recommend that after you upload the SVG, inject it into your web page. It's easy to manipulate in the DOM.

Next you will need to traverse over the SVG elements. Maker.js can [import the data from the `d` attribute](https://maker.js.org/docs/importing/#content) of `<path>` and `<polygon>` elements. If you encounter a `<circle>` add a Maker.js circle path. If it is an `<ellipse>` or `<rect>`, use the corresponding Maker.js model. You may also wish to just normalize everything as a `<path>` using a library like [shape2path](https://www.npmjs.com/package/shape2path).

While traversing, you may need to consider if any of your SVG elements have `transform` attributes. This is an advanced scenario - during your initial coding phase I would avoid documents that use it. Keep things simple and we can talk about that later 😉.

Once you have everything imported as a Maker.js model, you can [find chains](https://maker.js.org/docs/working-with-chains/#content). You can also detect when [chains are nested](https://maker.js.org/docs/working-with-chains/#Find%20multiple%20chains). For each chain, you can call [makerjs.chain.dogbone](https://maker.js.org/docs/api/modules/makerjs.chain.html#dogbone) with your radius. [Here is an example](https://maker.js.org/playground/?script=dogbone-polygon) of finding a chain from a polygon, then adding dogbone fillets.

Hope that sets you off to a start, let me know if you have any questions!
-Dan

3kmch, 2018-05-23:
Hi Dan

Thank you for your answer with lots of hints.
And yes I would like to build a little browser app. :)

I will let you know when I've got a (partial) solution or "unsolvable" problems arise.

Kurt

vespakoen, 2018-12-13:
You might be interested in my MakerCAM project!

https://github.com/makercam

It also has a "quick n dirty" SVG import as described above, which can be found here:

https://github.com/makercam/makercam/blob/master/src/make.ts#L170-L228

Hope it helps =)

danmarshall, 2019-01-10:
Hi Kurt, I'm going to close this issue for now. Please feel free to reopen it if you have any further questions.

closed 2019-01-10 by danmarshall