microsoft/maker.js #338: DXF Export: DXF Requires group a 66 group in line[...]

danton721, 2018-06-01:
Hello, I'm having this problem while exporting with the options to polylines set to true. My code is:

`<!DOCTYPE html>
<html lang='en'>
  <head>
      <meta charset='UTF-8'>
      <title>DXF Browser Example</title>
      <script src='https://cdnjs.cloudflare.com/ajax/libs/require.js/2.3.2/require.min.js'></script>
      <script src="https://maker.js.org/target/js/browser.maker.js" type="text/javascript"></script>
      <script src="util/filesaver.js"></script>
  </head>
  <body>
    <input type='file' id='file'/>
    <div id='dxfresult'></div>
    <script>
    // DXF to SVG
      var fileInput = document.getElementById('file')
      window.requirejs(['../dist/dxf'], function(dxf) {
        fileInput.addEventListener('change', function(event) {
          var file = event.target.files[0]
          var reader = new FileReader()
          reader.onload = function(e) {
            if (e.target.readyState === 2) {
              var dxfContents = e.target.result
              var parsed = dxf.parseString(dxfContents)
			  const svg = dxf.toSVG(parsed)
			  document.getElementById('dxfresult').innerHTML = svg
			  
			  
			  // processing svg and so on
			  // TODO
			  
			  // Save SVG to DXF
			  var makerjs = require('makerjs')			  
			  var paths = document.getElementById('dxfresult').getElementsByTagName('svg')[0].getElementsByTagName('path')
			  var path = []
			  for(i = 0; i < paths.length; i++){
				  path.push(paths[i].getAttribute('d'))
			  }
			  var model = makerjs.importer.fromSVGPathData(path)
			  
			  var dxfexp = makerjs.exporter.toDXF(model, { usePOLYLINE: true })
			  
			  var blob = new Blob([dxfexp], {type: "dxf/dxf;charset=utf-8"});
			  saveAs(blob, 'output.dxf')
            }
          }
          reader.readAsBinaryString(file)
        })
      });
    </script>
  </body>
</html>`

I know my DXF to SVG converting process only create lines yet (looking forward and studying to implement arcs on both import and export), but I don't think the error does come from this process...

Samples of dxf files I used are in link bellow, and _dxf.dxf_ is used on input and _output.dxf_, like the name suggest, is the output.
https://we.tl/qnIdUrtkuv

danmarshall, 2018-06-01:
Hi @danton721 - thanks for providing the sample. I tried running it but I don't have `dist/dxf.js` and `util/filesaver.js`. Maybe you can save me a step and just paste the SVG, and we can continue from there?

danton721, 2018-06-01:
Sorry @danmarshall, late night, wasn't thinking straight... Updated the code now:

```
<!DOCTYPE html>
<html lang='en'>
  <head>
      <meta charset='UTF-8'>
      <title>DXF Browser Example</title>
      <script src='https://cdnjs.cloudflare.com/ajax/libs/require.js/2.3.2/require.min.js'></script>
      <script src='https://cdnjs.cloudflare.com/ajax/libs/require.js/2.3.2/require.min.js'></script>
	  <script src="https://maker.js.org/target/js/browser.maker.js" type="text/javascript"></script>
      <script>
      </script>
  </head>
  <body>
    <div id='dxfresult'>
	    <svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" version="1.1" preserveAspectRatio="xMinYMin meet" viewBox="6.938141722158564 -13.890155630852316 9.058674004302436 3.7801055235421703" width="100%" height="100%">
	  		<path fill="none" stroke="#4f0000" stroke-width="0.1%" d="M12.50214842447917,-10.31716548014323L12.519324855980965,-10.314548856757646L12.5510055277321,-10.307525420921477L12.57134995033689,-10.30144673989569L12.584720822823007,-10.29680736865639L12.622524686404443,-10.281148495650857L12.64462474283854,-10.27012568522135L12.70023784583053,-10.2402905085664L12.706107799555882,-10.237337718078662L12.715012802434346,-10.23364914511343L12.724205399404829,-10.230750730421985L12.733615629253741,-10.228664532690237L12.743171874449882,-10.227406429142608L12.75099974283854,-10.22700068522135L12.751452298476247,-10.226996995145083L12.753871118207453,-10.227102603095274L12.756271529259982,-10.227418623205574L12.758635263056314,-10.227942650372132L12.760944330155,-10.228670696431788L12.763181157161165,-10.229597220514423L12.765328720470661,-10.23071517121232L12.767370675829966,-10.232016040245629L12.769291482725828,-10.23348992721552L12.771076522657959,-10.2351256149522L12.772712210394637,-10.23691065488433L12.77418609736453,-10.238831461780192L12.775486966397837,-10.240873417139497L12.776604917095735,-10.243020980448993L12.77753144117837,-10.245257807455157L12.778259487238026,-10.247566874553844L12.778783514404584,-10.249930608350176L12.779099534514884,-10.252331019402705L12.779205142465075,-10.254749839133911L12.77913549978812,-10.25671470925543L12.77900687654053,-10.258338177334078L12.777876688188464,-10.266922810157437L12.776002600494355,-10.275376273455862L12.773398876397161,-10.283634231269312L12.770085331809431,-10.291633835552537L12.766087184806253,-10.299314204486645L12.76244310221354,-10.30514912272135L12.76008666537032,-10.30857955214186L12.747767173145716,-10.324634626989482L12.734095268335167,-10.339554892935043L12.7335161017056,-10.34013202178184L12.707139159492534,-10.364216657225636L12.68681614908854,-10.38024482584635L12.67983932505566,-10.38526960593378L12.6560497761906,-10.400425220033737L12.631029854937552,-10.413449766645474L12.604969978004695,-10.424244121104836L12.58388244520878,-10.43110998633827L12.573756484373183,-10.433924763620622L12.52579395229635,-10.444557800988898L12.477087201864858,-10.450970166332569L12.428006920859431,-10.45311305767948L12.38789036783854,-10.45168232584635L12.36882467510744,-10.450167575009566L12.2764342546898,-10.43845004399115L12.271207893238655,-10.437787204141083L12.265302572919538,-10.437529372287774L12.25939725260042,-10.437787204141083L12.25353687533466,-10.438558737444925L12.24832786783854,-10.43969013834635L12.248142893275652,-10.43973885311571L12.24628685506337,-10.440324059716156L12.244488883752762,-10.441068803817767L12.242762662991145,-10.441967417468243L12.241121330360839,-10.443013061675105L12.23957737739422,-10.444197778454562L12.23814255450563,-10.44551255139657L12.236827781563623,-10.446947374285159L12.235643064784165,-10.448491327251778L12.23501355511015,-10.44944109551247L12.230880793463522,-10.456309236647913L12.221956381436334,-10.473452872645867L12.22159349283854,-10.47423506022135L12.2214829405035,-10.47447335005309L12.218764159603646,-10.479696079568862L12.21560053361582,-10.484661977653346L12.21384940504577,-10.48704961278014L12.213262329245918,-10.487780913676174L12.21198847190944,-10.489171085016624L12.21059830056899,-10.490444942353102L12.209102395267879,-10.491592790862105L12.207512140748701,-10.492605894723388L12.20583963980857,-10.493476543604837L12.204097621189522,-10.494198111342754L12.202299342705109,-10.494765106370956L12.200458490340427,-10.49517321351492L12.198589074093508,-10.495419326832858L12.196705321350777,-10.495501573253812L12.19644066080729,-10.49549995279948L12.189960325021179,-10.495177251388746L12.182478079717457,-10.49419219510695L12.175110159927955,-10.492558766614836L12.167912639971144,-10.490289397289521L12.16157174252597,-10.48769272194859L12.14195189404097,-10.479658153152247L12.12814891276041,-10.47505512858073L12.124331068152623,-10.473964522002982L12.11405190880094,-10.471685687238256L12.103613251217434,-10.47031141189593L12.09309453988935,-10.469852155041162L12.082575828561266,-10.47031141189593L12.07213717097776,-10.471685687238256L12.06521506521574,-10.47311642379846L12.054652492616139,-10.475784864625389L12.023058317564105,-10.485746469754925L12.01898216471354,-10.48725654459635L11.997819721427662,-10.495096504077914L11.96094771348806,-10.50810422039815L11.92899974283854,-10.51800068522135L11.883022754394352,-10.532744121700059L11.86962262146089,-10.53779651203788L11.861077552220845,-10.541451603290243L11.849887623641427,-10.547276711406717L11.839247967743063,-10.55405491976479L11.829239558731537,-10.561734642106064L11.82859349283854,-10.56227998209635L11.824311169044487,-10.56622710 [... cut ...]
		var paths = document.getElementById('dxfresult').getElementsByTagName('svg')[0].getElementsByTagName('path')
		
		// Append all SVG Objects "d" path to array of paths
		var path = []
		for(i = 0; i < paths.length; i++){
			path.push(paths[i].getAttribute('d'))
		}
		
		// Import path to a makerjs model
		var model = makerjs.importer.fromSVGPathData(path)
		
		// To DXF polyline
		var dxfexp = makerjs.exporter.toDXF(model, { usePOLYLINE: true })
		
		//Save to file
		download(dxfexp, "output.dxf", "dxf")
    }
    </script>
  </body>
</html>
```

danmarshall, 2018-06-01:
Hi @danton721 - your drawing has some extremely short lines (one sample was 0.000672 😄 ) which are being [overlooked by the chain algorithm](https://github.com/Microsoft/maker.js/blob/8f6945ab/src/core/chain.ts#L206), whose [default is .005](https://github.com/Microsoft/maker.js/blob/8f6945ab/src/core/chain.ts#L155). Looks like this is one option I [neglected to expose in the DXF exporter](https://github.com/Microsoft/maker.js/blob/8f6945ab/src/core/dxf.ts#L264).

I've manually patched a version for you here: https://gist.github.com/danmarshall/4da6258bf22e9561363368f577e6b305

Seems to work when you add the `pointMatchingDistance` option:
```js
// To DXF polyline
var dxfexp = makerjs.exporter.toDXF(model, { usePOLYLINE: true, pointMatchingDistance: .0001 })
```
I'll be adding this option to the published version soon. Thanks for pointing this out! 👍

danton721, 2018-06-02:
Thanks @danmarshall will test out later tomorrow! 😄

danmarshall, 2018-06-02:
I'm going to re-open this issue as a reminder to check in the fix. 👍

PR #359 (2018-08-19): pointMatchingDistance in Dxf chain
fixes #338
closed 2018-08-19 by danmarshall