microsoft/maker.js #342: Flex section: cloneToBrick

3kmch, 2018-07-02:
Is there already a function which helps to add flex sections? At the moment I'm using cloneToBrick which works fine. But from the point of material waste I would like to prevent all unnecessary cuts.  

[image]

See also https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/b4568769884592d8253f64c62de5c1f8/raw/4ddd1ddcdaa2d128fa93e3576e9f511129c7d017/beerkini.js

danmarshall, 2018-07-04:
I'd say keep using cloneToBrick. To clip the paths outside, you can use [makerjs.model.combine](https://maker.js.org/docs/api/modules/makerjs.model.html#combine) with a special combination of flags:

```js
makerjs.model.combine(bottom, wall, true, true, true, false, {trimDeadEnds: false});
```

danmarshall, 2018-07-04:
just do this prior to calling 
```js
  makerjs.model.combineUnion(bottom, top); // Combine bottom and top
```
so that `bottom` is still a closed geometry.

3kmch, 2018-07-04:
Thank you very much Dan, the problem is solved. :)

Feel free to add it to the gallery

https://www.thingiverse.com/thing:2989232

closed 2018-07-04 by 3kmch