microsoft/maker.js #349: Wrapping text generated from the Text model

gpmcadam, 2018-08-01:
Is there a way when converting text to an SVG using the Text model to word-wrap the text at a certain width, rather than the output being the entire sentence spanning continuously?

danmarshall, 2018-08-01:
Hi @gpmcadam - it would be nice if the underlying library [opentype.js](https://opentype.js.org) provided this. I've opened [an issue in their repo](https://github.com/nodebox/opentype.js/issues/352).

You could also do it yourself, by measuring each glyph in a Text model. Things to watch out for would be:
* What is the line height (this could be passed in)
* Where should glyphs be left aligned (this might require more work.)

Another approach, if you  are in a browser environment, is to load the same font into your web page, and render each character as a `span` into a DOM element which had a `max-width` setting. Then you could get the location of each `span`.

danmarshall, 2018-08-02:
According to the OpenType team, guidance would be to try to use the [opentype-layout](https://github.com/Jam3/opentype-layout) project. As a step to support opentype-layout, I've exposed a new function `makerjs.models.Text.glyphToModel()` which accepts a glyph and returns a model.

We should then be able to call the [computeLayout](https://github.com/Jam3/opentype-layout#layout--computelayoutfont-text-opt), and from the result then convert glyphs to models. I haven't tried it yet, but I will try to put a proof together when I get some time.

danmarshall, 2018-08-18:
Hi @gpmcadam , finally got some time to try opentype-layout, it works well! 
Code is here: https://github.com/danmarshall/makerjs-wrap-text
Live demo: https://maker.js.org/playground/?script=makerjs-wrap-text

closed 2018-08-18 by danmarshall