microsoft/maker.js #352: Not working with vue.js

icodk, 2018-08-11:
Managed(?) to install it for vue cli project but get an error trying to generate svg and use it
When I log the makerjs object I get:
`{version: "0.10.3", environmentTypes: {…}, environment: "browser", unitType: {…}, splitDecimal: ƒ, …}`
What I did so far:
1.` npm install makerjs --save`
2. In main.js:
```
import makerjs from 'makerjs'
global.makerjs=makerjs;
```
3. In a component, in one of the  computed functions:
```
var circle = {
            type: 'circle',
            origin: [0, 0],
            radius: 50
           };
var svg = makerjs.exporter.toSVG(circle);
console.log(svg);
```
4 Get the following error:
```
vue.esm.js?efeb:591 [Vue warn]: Error in render: "TypeError: kdbush is not a function"

found in

---> <Rectangle> at src\components\Rectangle.vue
       <App> at src\App.vue
         <Root>
warn @ vue.esm.js?efeb:591
logError @ vue.esm.js?efeb:1737
globalHandleError @ vue.esm.js?efeb:1732
handleError @ vue.esm.js?efeb:1721
Vue._render @ vue.esm.js?efeb:4546
updateComponent @ vue.esm.js?efeb:2788
get @ vue.esm.js?efeb:3142
Watcher @ vue.esm.js?efeb:3131
mountComponent @ vue.esm.js?efeb:2795
Vue.$mount @ vue.esm.js?efeb:8540
Vue.$mount @ vue.esm.js?efeb:10939
init @ vue.esm.js?efeb:4137
createComponent @ vue.esm.js?efeb:5608
createElm @ vue.esm.js?efeb:5555
updateChildren @ vue.esm.js?efeb:5845
patchVnode @ vue.esm.js?efeb:5936
patch @ vue.esm.js?efeb:6096
Vue._update @ vue.esm.js?efeb:2670
updateComponent @ vue.esm.js?efeb:2788
get @ vue.esm.js?efeb:3142
run @ vue.esm.js?efeb:3219
flushSchedulerQueue @ vue.esm.js?efeb:2981
(anonymous) @ vue.esm.js?efeb:1837
flushCallbacks @ vue.esm.js?efeb:1758
Promise.then (async)
microTimerFunc @ vue.esm.js?efeb:1806
nextTick @ vue.esm.js?efeb:1850
queueWatcher @ vue.esm.js?efeb:3068
update @ vue.esm.js?efeb:3209
Vue.$forceUpdate @ vue.esm.js?efeb:2691
(anonymous) @ index.js?737e:233
(anonymous) @ index.js?737e:231
(anonymous) @ index.js?737e:116
(anonymous) @ Rectangle.vue?99d4:39
(anonymous) @ Rectangle.vue?99d4:44
(anonymous) @ build.js:1029
__webpack_require__ @ build.js:679
hotApply @ build.js:608
(anonymous) @ build.js:290
Promise.then (async)
hotUpdateDownloaded @ build.js:289
hotAddUpdateChunk @ build.js:266
webpackHotUpdateCallback @ build.js:8
(anonymous) @ 0.450991b3595aa743ceee.hot-update.js:1
vue.esm.js?efeb:1741 TypeError: kdbush is not a function
    at PointGraph.mergeNearestSinglePoints (index.js?196a:2827)
    at _loop_2 (index.js?196a:5788)
    at Object.findChains (index.js?196a:5817)
    at getPathDataByLayer (index.js?196a:6956)
    at Object.toSVG (index.js?196a:7193)
    at VueComponent.drawProject (Rectangle.vue?d21f:212)
    at VueComponent.area (Rectangle.vue?d21f:160)
    at Watcher.get (vue.esm.js?efeb:3142)
    at Watcher.evaluate (vue.esm.js?efeb:3249)
    at VueComponent.computedGetter [as area] (vue.esm.js?efeb:3507)

```

danmarshall, 2018-08-11:
Looks like your build is not bringing in dependencies? Perhaps you can share your project.

icodk, 2018-08-11:
https://github.com/icodk/makerjs_vue.git

icodk, 2018-08-11:
I manually changed the kdbush dependency in makerjs from 2.0.1 to 1.0.1 (which is before kdbush was converted to ES6 modules) and that worked.

danmarshall, 2018-08-11:
🤔 Hmmm not sure why you'd need to do that... maybe some configuration of your bundler? Anyways glad you got it working!

icodk, 2018-08-11:
I think that the problem is that in 2.0.1 according to ES6,  kdbush  will return  an object by default
but makerjs expecting a function

danmarshall, 2018-08-12:
Can you share your webpack config?

icodk, 2018-08-12:
[webpack.config.zip](https://github.com/Microsoft/maker.js/files/2280807/webpack.config.zip)

icodk, 2018-08-12:
[makerjs.zip](https://github.com/Microsoft/maker.js/files/2280808/makerjs.zip)

icodk, 2018-08-12:
Hi again
I finally managed to get it working without fidling with the build system
Here is my way of doing it, let meknow if you mean there is a better way of doing it:

index.html
```
<html lang="en">
  <head>
    <meta charset="utf-8">
    <title>My App</title>
         
      <script src="https://cdn.jsdelivr.net/npm/makerjs@0/target/js/browser.maker.js"></script>
  
    <script>
    var makerjs=require('makerjs');
    </script>
  </head>
  <body>
    <div id="app"></div>
    <script src="/dist/build.js"></script>
  </body>
</html>
```
Vue component, somewhere in the template:
`<div id="drawing" ></div>`

in the vue component the updateted and mounted

 ```
updated:function(){
     console.log("Updated");
     this.drawProject();
   },
   mounted:function(){
     console.log("mounted");
     this.drawProject();

```
The drawProject function:
  ```
var circle = {
            type: 'circle',
            origin: [0, 0],
            radius: 5,
           };
          var svg = makerjs.exporter.toSVG(circle);
       
          var el=document.getElementById("drawing");
          if(el){
            el.innerHTML=svg;
          }
```

danmarshall, 2018-08-13:
Hi @icodk that's a good workaround. 
I was able to reproduce your original issue in a vue project. I'm not sure why it gets bundled the way it does, bu I might be able to add some code which can work with it.

danmarshall, 2018-08-15:
Hi @icodk I'm not a Vue developer, but I did a lot of searching through issues in the vue cli project, and vue forum. It appears that the default babel/webpack stack has issues mixing CommonJs with ES5/6. 
Maker.js currently ships a CommonJs module. For another workaround, you may want to try converting it with https://github.com/nolanlawson/cjs-to-es6 in a post-install step. I do plan on publishing an ES6 version but that will not be soon.

danielsdeboer, 2018-09-14:
I've had the same issue since 0.10, reverting back to 0.9 fixes the issue. Not super ideal but it'll do for now.

danmarshall, 2018-09-14:
@danielsdeboer I just published 0.11.2 which hopefully fixes this. Can you reinstall and try again?

danielsdeboer, 2018-09-15:
Yup, that fixed it. Thanks!

PR #366 (2018-09-14): look for default export
Packagers might use a "default" export. Hopefully this fixes #352
closed 2018-09-15 by danmarshall