microsoft/maker.js #372: Feature request: Hatch or fill  function

icodk, 2018-10-25:
Is it possible to draw a filled object 
... and export it to DXF ?
If not natively available, what will it take to simulate it ?
May be on a separate layer with different line thickness or expandingseveral times

danmarshall, 2018-10-25:
Should be possible with Boolean operations, and some mental gymnastics 🤔 !

Start with your model and a filler model. Let's use a smile and a brick wall:
[image]

Then subtract the wall from the smile. (This ruins the smile, we will deal with that afterwards):
[image]

FYI - here is what the wall looks like by itself: (not a closed geometry)
[image]

So we create a new smile! You can also do this by [cloning](https://maker.js.org/docs/intermediate-drawing/#Cloning) (prior to calling combineSubtraction). Smile1 was a sacrificial model.

[image]

Here's sample code:

```js
var m = require('makerjs');
var Smile = require('makerjs-smile');

function hatch() {

	var brick = new m.models.Rectangle(8, 6);
	var wall = m.layout.cloneToBrick(brick, 8, 7, 2);
    m.model.center(wall);
    
	var smile1 = new Smile();
  
    m.model.combineSubtraction(smile1, wall);
  
    var smile2 = new Smile();
	this.models = { wall, smile2 };
  
}

module.exports = hatch;
```
Or [click here to try it now](https://maker.js.org/playground/?script=https://gist.githubusercontent.com/danmarshall/e9a02b4cb8d94625c64fdedd2365f518/raw/382f30f1abd784363d77fa5fb6beb714a63f4a7a/hatch-fill.js)

icodk, 2018-10-26:
@danmarshall Thanks
Very interesting concept that I might be able to use instead of a solid fill

danmarshall, 2018-10-27:
Hi @icodk - you've inspired me to create a [separate module to do rasterizing](https://github.com/danmarshall/makerjs-raster).

[image]

[Here is a demo](https://maker.js.org/playground/?script=raster-smile) where you can refine the distance between the lines.

See [this line](https://github.com/danmarshall/makerjs-raster/blob/master/index.ts#L9) for the API; it will do the cloning for you, and return a model of horizontal lines.

icodk, 2018-10-28:
looks great
Thanks

closed 2018-10-29 by danmarshall

icodk, 2018-10-30:
A small optimization could be to check if the object  width is greater then height and to fill accordingly, in order to minimize number of path