microsoft/maker.js #375: Empty layers export to DXF

icodk, 2018-10-28:
Hi
I need to prepare some empty layers that will be exported to  DXF together with the rest of the model. Is there a way to create empty layers with names that will just be there when imported to ex. LibraCAD ,eventually with  more settings like font size etc.  ?

danmarshall, 2018-10-30:
I've marked this as an enhancement, so I won't get around to it until after some other work.
Meanwhile, I wonder if you can get by with this workaround: using a degenerate path, meaning either:
* a circle with radius 0
* a line with both ends on the same point

Currently, the dxf export code will create the layer table from the paths in the drawing. Then my question is if in your process, will LibraCAD ignore these "invisible" entities?

icodk, 2018-10-30:
I will try and let you know
 thanks

icodk, 2018-10-31:
it works by using Rectangle
```
var l1 = new makerjs.models.Rectangle(0,0);
l1.layer='layer1';
```
And then in the 
```
var model={
    models: {
        objects: otherobjects,
          l1:l1,
         }
     }
```