microsoft/maker.js #381: Offset when using makerjs.layout.childrenOnChain

firetunes, 2018-12-04:
First off, awesome library!

I'm emulating your example here: https://maker.js.org/playground/?script=text-on-chain. What I'd like to do is offer the ability to add an offset on either side of the text instead of having the text span the entire path. There are a few ways i could see doing it:

1) Be able to select a sub-section of the chain being used. Perhaps something like:
makerjs.chain.subChain(chain, start_percent_along_chain, end_percent_along_chain)

2) Be able to set the point of the first and last shapes on the chain, the other shapes will be evenly spaced.

3) Be able to set the point of the first shape and the spacing of the children along the path.

My preference would be the ability to get a subsection of the chain, as it seems a little more useful than the other two options.

danmarshall, 2018-12-04:
Hi @firetunes , thanks for the feedback 👍 
First off, I'd say take a look at the [source code for layout.childrenOnChain](https://github.com/Microsoft/maker.js/blob/master/packages/maker.js/src/core/layout.ts#L183) - the **contain** flag does almost what you want.
Secondly, it may be easier to add temporary construction lines than to create a new algorithm. Here's the same code using this technique:
```js
var makerjs = require('makerjs');

function TextOnChain(font, text, fontSize, baseline, reversed, svgPathData, showPath, offset, deleteOffset) {

    var textModel = new makerjs.models.Text(font, text, fontSize);

    var svgPath = makerjs.importer.fromSVGPathData(svgPathData);

    var chain = makerjs.model.findSingleChain(svgPath);

    var m = makerjs.measure.modelExtents(textModel);

    if (offset) {
        textModel.models.leftOffset = addOffsetLine(m, true, offset);
        textModel.models.rightOffset = addOffsetLine(m, false, offset);
    }

    makerjs.layout.childrenOnChain(textModel, chain, baseline, reversed, true);

    if (deleteOffset) {
        delete textModel.models.leftOffset;
        delete textModel.models.rightOffset;
    }

    this.models = {
        text: textModel
    };

    if (showPath) {
        this.models.svgPath = svgPath;
    }
}

function addOffsetLine(m, leftSide, size) {
    var x1, x2;
    if (leftSide) {
        x1 = m.low[0];
        x2 = x1 - size;
    } else {
        x1 = m.high[0];
        x2 = x1 + size;
    }
    return {
        paths: {
            offset: new makerjs.paths.Line([x1, 0], [x2, 0])
        }
    };
}

TextOnChain.metaParameters = [
    { title: "font", type: "font", value: '*' },
    { title: "text", type: "text", value: 'We go up, then we go down, then up again' },
    { title: "font size", type: "range", min: 1, max: 200, value: 42.5 },
    { title: "baseline", type: "range", min: -1, max: 2, step: 0.1, value: 0.5 },
    { title: "reversed", type: "bool", value: false },
    { title: "svg path data", type: "text", value: 'M 100 200 C 200 100 300 0 400 100 C 500 200 600 300 700 200 C 800 100 900 100 900 100' },
    { title: "show path", type: "bool", value: true },
    { title: "offset", type: "range", value: 20, min: 0, max: 100 },
    { title: "delete offset lines", type: "bool", value: false }
];

module.exports = TextOnChain;
```

firetunes, 2018-12-04:
Hi @danmarshall 

So was able to take your idea and i modified it as such:

```
var makerjs = require('makerjs');

function TextOnChain(font, text, fontSize, baseline, reversed, svgPathData, showPath, startpercent, endpercent) {
    var i;
    var divisions = 1000;
    var textModel = new makerjs.models.Text(font, text, fontSize);

    var svgPath = makerjs.importer.fromSVGPathData(svgPathData);

    var chain = makerjs.model.findSingleChain(svgPath);
    var points = makerjs.chain.toPoints(chain, chain.pathLength / divisions);
    makerjs.chain.reverse(chain);
    var lastpoints = makerjs.chain.toPoints(chain, chain.pathLength);
    points.push(lastpoints[0]);
    makerjs.chain.reverse(chain);
  
    var textpoints = [];
    for(i = startpercent * 10; i < endpercent * 10; i++) {
      textpoints.push(points[i]);
    }
    var textcurve = new makerjs.models.ConnectTheDots(false, textpoints);
    var newchain = makerjs.model.findSingleChain(textcurve);
    makerjs.chain.reverse(newchain);
  
    makerjs.layout.childrenOnChain(textModel, newchain, baseline, reversed, true);

    this.models = {
        text: textModel
    };
  
    if (showPath) {
        this.models.svgPath = svgPath;
    }
    this.models.textcurve = textcurve;
}

TextOnChain.metaParameters = [
    { title: "font", type: "font", value: '*' },
    { title: "text", type: "text", value: 'We go up, then we go down' },
    { title: "font size", type: "range", min: 1, max: 200, value: 42.5 },
    { title: "baseline", type: "range", min: -1, max: 2, step: 0.1, value: 0.5 },
    { title: "reversed", type: "bool", value: false },
    { title: "svg path data", type: "text", value: 'M 100 200 C 200 100 300 0 400 100 C 500 200 600 300 700 200 C 800 100 900 100 900 100' },
    { title: "show path", type: "bool", value: false },
    { title: "start percent", type: "range", value: 20, min: 0, max: 100, step: 0.1 },
    { title: "end percent", type: "range", value: 80, min: 0, max: 100, step: 0.1 }
];

module.exports = TextOnChain;
```

So effectively i get an approximated "sub-chain" from the above and i can do the text on that sub-chain.

Thanks for your help in a starting direction.

danmarshall, 2018-12-04:
@firetunes nice, I like that approach!

firetunes, 2018-12-04:
@danmarshall 

is there a cleaner way to get the endpoint of a chain? As you can see above I just reversed the chain, got another point array and just grabbed the first point from it. Not sure if there was a built in function to get the start / end points of a chain.

Thanks.

danmarshall, 2018-12-04:
See the documentation for [chain links](https://maker.js.org/docs/working-with-chains/#Chain%20links).
There is an endpoints array, and a reversed flag. Which endpoint you need depends on if the link is at the beginning or end of the chain, and if the link is reversed. There is a [function in the source code](https://github.com/Microsoft/maker.js/blob/master/packages/maker.js/src/core/chain.ts#L113), but it isn't exposed. Perhaps it could be.

firetunes, 2018-12-05:
Well for now I'll leave it as is. Seems to work well. Thanks for your help!

closed 2018-12-05 by firetunes